coolsnowwolf/lede #3216: 请教下Ubuntu的neteasemusic怎么劫持

zwillhill, 2020-02-20:
cp CA.crt /usr/share/ca-certificates/x-net.crt

拷贝证书到/usr/share/ca-certificates/下

sudo chmod a+r /usr/share/ca-certificates/x-net.crt

修改权限

sudo dpkg-reconfigure ca-certificates 

安装证书
————————————————
按照上面操作安装了证书劫持无效，显示网络无法连接，Ubuntu版本18.04，Neteasemusic客户端版本1.2，chrome网页版显示证书错误，忽略后也是无法解锁

nivalxer, 2020-02-20:
Chrome等网页在Ubuntu下要信任证书还需要使用certutil导入到stores中
参考：https://thomas-leister.de/en/how-to-import-ca-root-certificate/
网页已信任，未找到客户端下载链接，未测试客户端
[image]

zwillhill, 2020-02-20:
> Chrome等网页在Ubuntu下要信任证书还需要使用certutil导入到stores中
> 参考：https://thomas-leister.de/en/how-to-import-ca-root-certificate/
> 网页已信任，未找到客户端下载链接，未测试客户端
> [image]

谢谢，按链接里操作，chrome证书没问题了。客户端还是显示网络错误
1.2.1版本（和深度合作开发，应该是以前的官方版）http://zitian.jb51.net:81/201906/tools/netease-cloud-music_ubuntu_jb51.deb
低版本好像装不上，这个能装，但是没有代理设置功能。

nivalxer, 2020-02-21:
> 谢谢，按链接里操作，chrome证书没问题了。客户端还是显示网络错误
> 1.2.1版本（和深度合作开发，应该是以前的官方版）http://zitian.jb51.net:81/201906/tools/netease-cloud-music_ubuntu_jb51.deb
> 低版本好像装不上，这个能装，但是没有代理设置功能。

用你发的客户端试了一下，正常。
[image]
[image]

zwillhill, 2020-02-21:
> 用你发的客户端试了一下，正常。
> [image]
> [image]

感谢测试！难道还是证书的问题？我再研究研究

shi429101906, 2020-02-23:
> 用你发的客户端试了一下，正常。
> [image]
> [image]

大佬，安卓上的也无法劫持， 这个有没有解决办法呢

zwillhill, 2020-02-23:
> 大佬，安卓上的也无法劫持， 这个有没有解决办法呢

安卓直接下载生成的证书安装就可以啦，

shi429101906, 2020-02-24:
> 安卓直接下载生成的证书安装就可以啦，

安装证书没有卵用， 安卓和ubuntu18一个样， 用了博客里写的加上代理， 也和上面的问题一样， 无法连接。  现在可以正常使用的只有win10的uwp版><

nivalxer, 2020-02-24:
> 安装证书没有卵用， 安卓和ubuntu18一个样， 用了博客里写的加上代理， 也和上面的问题一样， 无法连接。 现在可以正常使用的只有win10的uwp版><

没有安卓机可以测试，能提供一下安卓的版本号么？
空了我试试x86的模拟器看看。

zwillhill, 2020-02-24:
> 没有安卓机可以测试，能提供一下安卓的版本号么？
> 空了我试试x86的模拟器看看。

大佬方便写个导入Ubuntu证书的简易教程不？我是搞不定Ubuntu的证书了，网上搜了好多教程来来回回弄好几遍啦

nivalxer, 2020-02-24:
> https://thomas-leister.de/en/how-to-import-ca-root-certificate/

就是我之前回复的这个：https://thomas-leister.de/en/how-to-import-ca-root-certificate/
不过看反馈好像没成功，我这边虚拟机是OK的。

shi429101906, 2020-02-24:
@nivalxer 我使用的是小米mix 2s, 系统版本号这两天刚更新, MUI11.0.1, 之前的版本10， 手机和ubuntu都安装了证书不起作用， https://www.maxlicheng.com/github/232.html， 使用这里的自定义代理， 网易将无法上网。 openwrt的版本是esir2月1日编译的高大全版

closed 2020-03-06 by coolsnowwolf