coolsnowwolf/lede #3222: 编译出现runc这个错误，怎么解决呢，求教

zhubang, 2020-02-20:
make[3]: Entering directory '/home/bang/lede/feeds/packages/utils/runc'
mkdir -p /home/bang/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/bang/lede/tmp/.root-copy.flock -c 'cp -fpR /home/bang/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.pkgdir/runc/. /home/bang/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/bang/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.runc_installed
mkdir -p /home/bang/lede/bin/targets/x86/64/packages /home/bang/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/ipkg-x86_64/runc/CONTROL /home/bang/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/bang/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/ipkg-x86_64/runc/usr/sbin/
install -m0755 /home/bang/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc/runc /home/bang/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/ipkg-x86_64/runc/usr/sbin/
install: cannot stat '/home/bang/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc/runc': No such file or directory
make[3]: *** [Makefile:94: /home/bang/lede/bin/packages/x86_64/packages/runc_1.0.0-rc8+91-3e425f80-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/bang/lede/feeds/packages/utils/runc'
time: package/feeds/packages/runc/compile#0.13#0.26#0.80
make[2]: *** [package/Makefile:109: package/feeds/packages/runc/compile] Error 2
make[2]: Leaving directory '/home/bang/lede'
make[1]: *** [package/Makefile:103: /home/bang/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bang/lede'
make: *** [/home/bang/lede/include/toplevel.mk:218：world] 错误 2

zhg-em, 2020-02-20:
make clean
然后重新编译

hyird, 2020-02-21:
rm -rf /home/bang/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/

zhubang, 2020-02-21:
> make clean
> 然后重新编译

没用

coolsnowwolf, 2020-02-21:
删除 dl 里面的 runc，重新下载一次

zhubang, 2020-02-21:
> rm -rf /home/bang/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/

试了可以用

zhubang, 2020-02-21:
> 删除 dl 里面的 runc，重新下载一次

感谢大雕回复，试了楼上老哥的方法可以用，rm -rf /home/bang/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/

feizaibufei, 2020-02-22:
每次编译都要删除runc-1.0.0-rc8+91-3e425f80这个目录

closed 2020-02-22 by coolsnowwolf