coolsnowwolf/lede #3232: Transmission无法正常使用

SacredOath, 2020-02-21:
设备K3,配置文件目录为挂载硬盘/mnt/sda1
Transmission可以启动,但启动后会报错,而且添加种子下载几十M后就会自动停止
Fri Feb 21 09:15:09 2020 daemon.err transmission-daemon[26230]: [2020-02-21 09:15:09.680] Couldn't save file "/mnt/sda1/transmission/settings.json": File exists (variant.c:1271)
Fri Feb 21 09:16:37 2020 daemon.err transmission-daemon[26230]: [2020-02-21 09:16:37.901] Couldn't save file "/mnt/sda1/transmission/settings.json": File exists (variant.c:1271)
Fri Feb 21 09:16:44 2020 user.notice transmission: Starting with 256398000 virt mem
Fri Feb 21 09:16:45 2020 daemon.err transmission-daemon[30627]: [2020-02-21 09:16:45.983] Couldn't save temporary file "/mnt/sda1/transmission/settings.json.tmp.XXXXXX": Permission denied (variant.c:1280)

EnnawYang, 2020-02-21:
无法写入，检查权限

SacredOath, 2020-02-21:
> 无法写入，检查权限

哪里的权限?我在Transmission设置界面用户组分别换了root,transmission,aria2都不行

SacredOath, 2020-02-21:
> 无法写入，检查权限

其实已经写入了,我用WinSCP打开能看到生成对应的settings.json文件,就是无法正常使用

coolsnowwolf, 2020-02-21:
用 root 用户执行

SacredOath, 2020-02-21:
> 用 root 用户执行

我把配置文件目录更改为路由内部存储后,一切正常了.我感觉是那个最新NTFS内核的问题?
因为旧版固件配置文件放在移动硬盘也一切正常.

xukaiKnight, 2020-02-22:
我也遇到了这个问题，配置目录如果是挂载的NTFS硬盘，自动生成的settings.json所属的group不是root，但是刷回旧版本就可以用。

closed 2020-02-22 by coolsnowwolf

Kisoul, 2021-09-05:
> 我也遇到了这个问题，配置目录如果是挂载的NTFS硬盘，自动生成的settings.json所属的group不是root，但是刷回旧版本就可以用。

你们解决了吗？我开始以为是NTFS的问题，后面我用了ext3和ext4，发现都不行，还是有这个问题。现在没办法退回到4.19.108的内核了（没用用antfs用的是ntfs-3g），话说老的版本编译的时候automount和antfs没有关联，现在强制关联了，老版本用的是automount+ntfs-3g，现在不行了，不知道是不是这个原因。

xukaiKnight, 2021-09-06:
没有解决，按上面说的，把config文件夹放内部存储，下载目录放挂在的硬盘

Kisoul, 2021-09-06:
> 没有解决，按上面说的，把config文件夹放内部存储，下载目录放挂在的硬盘

好吧，老的版本非常完美，但是功能上有一些缺失，新编译的，就是transmission有问题