coolsnowwolf/lede #3233: 2月21号编译报错

alepennyq, 2020-02-21:
cc1: some warnings being treated as errors
make[6]: *** [/home/alfisti/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/ioctl.o] Error 1
make[5]: *** [_module_/home/alfisti/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04] Error 2
make[5]: Leaving directory `/home/alfisti/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.101'
make[4]: *** [build] Error 2
make[4]: Leaving directory `/home/alfisti/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04'
make[3]: *** [/home/alfisti/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/.built] Error 2
make[3]: Leaving directory `/home/alfisti/lede/feeds/packages/utils/cryptodev-linux'
time: package/feeds/packages/cryptodev-linux/compile#0.82#0.16#2.08
make[2]: *** [package/feeds/packages/cryptodev-linux/compile] Error 2
make[2]: Leaving directory `/home/alfisti/lede'
make[1]: *** [/home/alfisti/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/alfisti/lede'
make: *** [world] Error 2

czjesda, 2020-02-21:
make clean后重新编译一下看看

alepennyq, 2020-02-21:
> make clean后重新编译一下看看

OK，正在重新编译中……

alepennyq, 2020-02-21:
> make clean后重新编译一下看看

还是一样的

Zxilly, 2020-02-21:
一样的，应该是这个的锅https://github.com/coolsnowwolf/lede/commit/9f8000a6d635aeb594505ca9538add537fae11f5

alepennyq, 2020-02-21:
> 一样的，应该是这个的锅[9f8000a](https://github.com/coolsnowwolf/lede/commit/9f8000a6d635aeb594505ca9538add537fae11f5)

应该是，等修复吧

Zxilly, 2020-02-21:
`git reset --hard 6561323807bfd690514e30d27d45002f89d68805`应该就行了

alepennyq, 2020-02-21:
> `git reset --hard 6561323807bfd690514e30d27d45002f89d68805`应该就行了

还是一样的：
cc1: some warnings being treated as errors
make[6]: *** [/home/alfisti/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/ioctl.o] Error 1
make[5]: *** [_module_/home/alfisti/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04] Error 2
make[5]: Leaving directory `/home/alfisti/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.101'
make[4]: *** [build] Error 2
make[4]: Leaving directory `/home/alfisti/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04'
make[3]: *** [/home/alfisti/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/.built] Error 2
make[3]: Leaving directory `/home/alfisti/lede/feeds/packages/utils/cryptodev-linux'
time: package/feeds/packages/cryptodev-linux/compile#0.85#0.12#1.91
make[2]: *** [package/feeds/packages/cryptodev-linux/compile] Error 2
make[2]: Leaving directory `/home/alfisti/lede'
make[1]: *** [/home/alfisti/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/alfisti/lede'
make: *** [world] Error 2

coolsnowwolf, 2020-02-21:
日志不全，更新 feeds 后看看

alepennyq, 2020-02-21:
> 日志不全，更新 feeds 后看看

alfisti@ubuntu:~/lede$ ./scripts/feeds update -a
Updating feed 'packages' from 'https://github.com/coolsnowwolf/packages' ...
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Compressing objects: 100% (1/1), done.
remote: Total 3 (delta 2), reused 3 (delta 2), pack-reused 0
Unpacking objects: 100% (3/3), done.
From https://github.com/coolsnowwolf/packages
   64ed8b5..c9c02d2  master     -> origin/master
Updating 64ed8b5..c9c02d2
Fast-forward
 utils/cryptodev-linux/Makefile | 74 --------------------------------------------------------------------------
 1 file changed, 74 deletions(-)
 delete mode 100644 utils/cryptodev-linux/Makefile
Create index file './feeds/packages.index' 
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'ldconfig-stub'... ok.
Collecting package info: done
Updating feed 'luci' from 'https://github.com/coolsnowwolf/luci' ...
Already up-to-date.
Create index file './feeds/luci.index' 
Collecting package info: done
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git;openwrt-19.07' ...
Already up-to-date.
Create index file './feeds/routing.index' 
Collecting package info: done

closed 2020-02-21 by alepennyq