coolsnowwolf/lede #3234: 建议更换README的推荐系统为Ubuntu 18.04 LTS

AmadeusGhost, 2020-02-21:
Ubuntu 14.04 LTS的5年支持已经过了，有些包确实过旧不适合编译，像编译sunxi的uboot还有切换到python3都会有问题，随着同步官方op源码，此类问题会越来越多，所以还是推荐Ubuntu 18.04 LTS好点，而且Actions也证明了Ubuntu 18.04可以稳定编译。

coolsnowwolf, 2020-02-21:
有道理

EstFoxs, 2020-02-21:
ubuntu 19 编译环境速度更快

hcym, 2020-02-21:
应该没问题，用了差不多一年了，年底再看看20.04

AmadeusGhost, 2020-02-21:
> ubuntu 19 编译环境速度更快

19不是LTS（

KFERMercer, 2020-02-21:
建议等20.04一步到位

EstFoxs, 2020-02-21:
> 建议等20.04一步到位

4月底才发布稳定版，还有两月

jyy5206, 2020-02-21:
用 Ubuntu 18.04.3 Desktop 版本编译了半年多了
一直很稳定

用的是笔记本电脑 Win10 + Ubuntu 双系统
缺省进入Win10
需要编译时切换进入 Ubuntu

亲测建议：
不要用 Win10 里面的 Hyper-V 虚拟机
慢的要死~

ZForest000, 2020-02-22:
用WIN10的也可以考虑WSL安装Ubuntu 18.04 LTS

xydche, 2020-02-22:
> 用 Ubuntu 18.04 LTS Desktop 版本编译了半年多了
> 一直很稳定
> 
> 用的是笔记本电脑 Win10 + Ubuntu 双系统
> 缺省进入Win10
> 需要编译时，重启切换到 Ubuntu
> 
> 亲测建议：
> 不要用 Win10 里面的 Hyper-V 虚拟机
> 慢的要死~

还行吧，hyper v感觉挺快的啊。主要是方便，编译的时候电脑该干麻干麻

rageq, 2020-02-26:
> 
> 
> 用 Ubuntu 18.04 LTS Desktop 版本编译了半年多了
> 一直很稳定
> 
> 用的是笔记本电脑 Win10 + Ubuntu 双系统
> 缺省进入Win10
> 需要编译时，重启切换到 Ubuntu
> 
> 亲测建议：
> 不要用 Win10 里面的 Hyper-V 虚拟机
> 慢的要死~

不妨试试win10自带的WLS2，比WLS提升不少，虚拟机，双系统什么的太臃肿了
WLS2还支持升级内核，1804可以升级到1910

amilys, 2020-02-29:
debian10 也可以

amilys, 2020-02-29:
> > 用 Ubuntu 18.04 LTS Desktop 版本编译了半年多了
> > 一直很稳定
> > 用的是笔记本电脑 Win10 + Ubuntu 双系统
> > 缺省进入Win10
> > 需要编译时，重启切换到 Ubuntu
> > 亲测建议：
> > 不要用 Win10 里面的 Hyper-V 虚拟机
> > 慢的要死~
> 
> 不妨试试win10自带的WLS2，比WLS提升不少，虚拟机，双系统什么的太臃肿了
> WLS2还支持升级内核，1804可以升级到1910

win的linux子系统文件不分大小写，容易出错

eskimokk, 2020-03-01:
> > > 用 Ubuntu 18.04 LTS Desktop 版本编译了半年多了
> > > 一直很稳定
> > > 用的是笔记本电脑 Win10 + Ubuntu 双系统
> > > 缺省进入Win10
> > > 需要编译时，重启切换到 Ubuntu
> > > 亲测建议：
> > > 不要用 Win10 里面的 Hyper-V 虚拟机
> > > 慢的要死~
> > 
> > 
> > 不妨试试win10自带的WLS2，比WLS提升不少，虚拟机，双系统什么的太臃肿了
> > WLS2还支持升级内核，1804可以升级到1910
> 
> win的linux子系统文件不分大小写，容易出错

亲测wsl（ubuntu 18.04 LTS）编译可用

closed 2020-03-05 by AmadeusGhost