coolsnowwolf/lede #3244: 同步最新源码后，云编译和本地编译出现同一错误。

haidaoyy, 2020-02-21:
只能看懂是个关于samba4的错误

Traceback (most recent call last):
  File "./buildtools/bin/waf", line 75, in <module>
    import Scripting
  File "/home/runner/work/MyLede/MyLede/openwrt/build_dir/target-mipsel_24kc_musl/samba-4.9.2/third_party/waf/wafadmin/Scripting.py", line 146
    except Utils.WafError, e:
                         ^
SyntaxError: invalid syntax
make[3]: *** [/home/runner/work/MyLede/MyLede/openwrt/build_dir/target-mipsel_24kc_musl/samba-4.9.2/.configured_4ba563fa1479974f3f1b9eb8e88b6916] Error 1
Makefile:394: recipe for target '/home/runner/work/MyLede/MyLede/openwrt/build_dir/target-mipsel_24kc_musl/samba-4.9.2/.configured_4ba563fa1479974f3f1b9eb8e88b6916' failed
make[3]: Leaving directory '/home/runner/work/MyLede/MyLede/openwrt/package/lean/samba4'
time: package/lean/samba4/compile#0.27#0.12#0.39
package/Makefile:111: recipe for target 'package/lean/samba4/compile' failed
make[2]: *** [package/lean/samba4/compile] Error 2
make[2]: Leaving directory '/home/runner/work/MyLede/MyLede/openwrt'
make[1]: *** [/home/runner/work/MyLede/MyLede/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
package/Makefile:107: recipe for target '/home/runner/work/MyLede/MyLede/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/home/runner/work/MyLede/MyLede/openwrt'
/home/runner/work/MyLede/MyLede/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
##[error]Process completed with exit code 2.

jwf8280031, 2020-02-21:
network取消samba36

haidaoyy, 2020-02-21:
> network取消samba36

你好，samba36一直是取消状态，且之前源码一直能编译samba4

今天执行了两步
1.   sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler 安装了个python3

2.   git pull 同步源码  和  ./scripts/feeds update -a && ./scripts/feeds install -a 更新feeds

之后编译出问题。

现在make clean，然后git reset --hard 30ffdb3 退回上个原本可编译源码后，执行编译
出现同样错误。

这个是什么问题呢

closed 2020-02-21 by haidaoyy

georgeent, 2020-04-21:
有同样的问题：IPQ4019编译固件，请各位指导依赖程序或文件
touch /data/exhome/wrt-worker/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/talloc-2.1.11/.prepared_d54269edde38d0ac1079d53a579a9cbb_6664517399ebbbc92a37c5bb081b5c53
rm -f /data/exhome/wrt-worker/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/talloc-2.1.11/.configured_*
rm -f /data/exhome/wrt-worker/lede-master/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.libtalloc_installed
(cd /data/exhome/wrt-worker/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/talloc-2.1.11; echo -e >cache.txt " \nChecking simple C program: OK \nrpath library support: OK \n-Wl,--version-script support: OK \nChecking getconf LFS_CFLAGS: NO \nChecking for large file support without additional flags: OK \nChecking for -D_LARGE_FILES: OK \nChecking correct behavior of strtoll: NO \nChecking for working strptime: OK \nChecking for C99 vsnprintf: OK \nChecking for HAVE_SHARED_MMAP: OK \nChecking for HAVE_MREMAP: OK \nChecking for HAVE_INCOHERENT_MMAP: NO \nChecking for HAVE_SECURE_MKSTEMP: OK \nChecking for HAVE_IFACE_GETIFADDRS: OK \nChecking for kernel change notify support: OK \nChecking for Linux kernel oplocks: OK \nChecking for kernel share modes: OK \nChecking if can we convert from CP850 to UCS-2LE: OK \nChecking if can we convert from UTF-8 to UCS-2LE: OK \nChecking whether we can use Linux thread-specific credentials with 32-bit system calls: OK \nChecking whether we can use Linux thread-specific credentials: OK \nChecking whether setreuid is available: OK \nChecking whether setresuid is available: OK \nChecking whether seteuid is available: OK \nChecking whether fcntl locking is available: OK \nChecking for the maximum value of the 'time_t' type: OK \nChecking whether the realpath function allows a NULL argument: OK \nChecking whether POSIX capabilities are available: OK \nChecking for ftruncate extend: OK \nvfs_fileid checking for statfs() and struct statfs.f_fsid: OK \ngetcwd takes a NULL argument: OK \nChecking value of NSIG: \"65\" \nChecking value of _NSIG: \"65\" \nChecking value of SIGRTMAX: \"64\" \nChecking value of SIGRTMIN: \"34\" \nChecking if toolchain accepts -fstack-protector: OK \n" " \nChecking uname machine type: \"arm\" \nChecking uname release type: \"4.19.108\" \nChecking uname sysname type: \"Linux\" \nChecking uname version type: \"4.19.108\" \n" ; AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/data/exhome/wrt-worker/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/talloc-2.1.11:talloc-2.1.11 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/data/exhome/wrt-worker/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/talloc-2.1.11:talloc-2.1.11 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/data/exhome/wrt-worker/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/talloc-2.1.11:talloc-2.1.11 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/data/exhome/wrt-worker/lede-master/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I/data/exhome/wrt-worker/lede-master/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/include -I/data/exhome/wrt-worker/lede-master/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.5.0_musl_eabi/usr/include -I/data/exhome/wrt-worker/lede-master/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.5.0_musl_eabi/include/fortify -I/data/exhome/wrt-worker/lede-master/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.5.0_musl_eabi/include " LDFLAGS="-L/data/exhome/wrt-worker/lede-master/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib -L/data/exhome/wrt-worker/lede-master/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lib -L/data/exhome/wrt-worker/lede-master/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.5.0_musl_eabi/usr/lib -L/data/exhome/wrt-worker/lede-master/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-7.5.0_musl_eabi/lib -znow -zrelro "  ./buildtools/bin/waf configure --prefix=/usr --sysconfdir=/etc --localstatedir=/var --with-libiconv="" --cross-compile --cross-answers=/data/exhome/wrt-worker/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/talloc-2.1.11/cache.txt --disable-rpath --disable-rpath-install )
Traceback (most recent call last):
  File "./buildtools/bin/waf", line 75, in <module>
    import Scripting
  File "/data/exhome/wrt-worker/lede-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/talloc-2.1.11/third_party/waf/wafadmin/Scripting.py", line 146
    except Utils.WafError, e:
                         ^
SyntaxError: invalid syntax

xiangfeigao, 2020-04-21:
我也想知道