coolsnowwolf/lede #3249: 以前SS的socks server可以指定使用一个非默认的服务器

wayzha, 2020-02-21:
以前SS的socks server可以指定使用一个非默认的服务器，为何这个功能被移除了呢。
能否可以加回来。
或者更好的是可以根据域名/ip规则来走不同的SS服务器

或者哪位可以提供一个workaround嘛？

coolsnowwolf, 2020-02-21:
因为老有人提为啥不能开 ss 服务器的 socks 之类，所以砍掉了

wayzha, 2020-02-21:
> 因为老有人提为啥不能开 ss 服务器的 socks 之类，所以砍掉了

谢谢极速回复。 但是如果有需求部分访问走特定SS，有没有什么好的workaround？

EnnawYang, 2020-02-21:
> 因为老有人提为啥不能开 ss 服务器的 socks 之类，所以砍掉了

我觉得socks可以加一个走全局还是走规则的选项，之前的socks是走全局的（但不能用v2，现在的是走规则突然有点不适应

coolsnowwolf, 2020-02-21:
主要是空间不足了，如果用那个 socks 8M ROM 的小路由器放不下了

wayzha, 2020-02-21:
> 主要是空间不足了，如果用那个 socks 8M ROM 的小路由器放不下了

可以做为一个optional的选项编译的时候选择嘛：）

DongfeiSay, 2020-02-21:
> 因为老有人提为啥不能开 ss 服务器的 socks 之类，所以砍掉了

socks server能否添加认证用户名密码的选择呢？

coolsnowwolf, 2020-02-22:
不是做了 v2ray server 的luci 吗？里面就有用户名密码选项

TrustAsia, 2020-02-22:
> 因为老有人提为啥不能开 ss 服务器的 socks 之类，所以砍掉了

请问如何恢复这个功能？ 出于地区等特殊原因我这里需要socks5+proxifier进行全局代理 希望能把这个功能加回来

coolsnowwolf, 2020-02-22:
不恢复了，之前也没啥人用

TrustAsia, 2020-02-22:
> 不恢复了，之前也没啥人用

请问如果我本地回滚的话需要怎么操作？应该回滚到哪一个commit?

closed 2020-02-22 by coolsnowwolf