coolsnowwolf/lede #3251: Python报错

Zxilly, 2020-02-21:
是因为python2和python3的兼容问题吗
```log
Enabling rpcd
Enabling shadowsocksr
Enabling sqm
Enabling sysctl
Enabling sysfixtime
Enabling sysntpd
Enabling system
Enabling uhttpd
Enabling umount
./etc/init.d/unblockneteasemusic-mini: line 8: uci: command not found
./etc/init.d/unblockneteasemusic-mini: line 9: uci: command not found
Enabling unblockneteasemusic-mini
Enabling urandom_seed
Enabling urngd
Enabling vsftpd
Enabling zerotier
sed -i "s/Installed-Time: .*/Installed-Time: 1582266658/" /home/zxilly/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/status
rm -rf /home/zxilly/lede/build_dir/target-mipsel_24kc_musl/root-ramips/tmp/*
rm -f /home/zxilly/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/lists/*
rm -f /home/zxilly/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/info/*.postinst*
rm -f /home/zxilly/lede/build_dir/target-mipsel_24kc_musl/root-ramips/var/lock/*.lock
rm -rf /home/zxilly/lede/build_dir/target-mipsel_24kc_musl/root-ramips/boot
rm -rf /home/zxilly/lede/tmp/mklibs-progs /home/zxilly/lede/tmp/mklibs-out
# first find all programs and add them to the mklibs list
find /home/zxilly/lede/staging_dir/target-mipsel_24kc_musl/root-ramips -type f -perm /100 -exec file -r -N -F '' {} + | awk ' /executable.*dynamically/ { print $1 }' > /home/zxilly/lede/tmp/mklibs-progs
# find all loadable objects that are not regular libraries and add them to the list as well
find /home/zxilly/lede/staging_dir/target-mipsel_24kc_musl/root-ramips -type f -name \*.so\* -exec file -r -N -F '' {} + | awk ' /shared object/ { print $1 }' > /home/zxilly/lede/tmp/mklibs-libs
mkdir -p /home/zxilly/lede/tmp/mklibs-out
/home/zxilly/lede/staging_dir/host/bin/mklibs -D -d /home/zxilly/lede/tmp/mklibs-out --sysroot /home/zxilly/lede/staging_dir/target-mipsel_24kc_musl/root-ramips `cat /home/zxilly/lede/tmp/mklibs-libs | sed 's:/*[^/]\+/*$::' | uniq | sed 's:^/home/zxilly/lede/staging_dir/target-mipsel_24kc_musl/root-ramips:-L :'` --ldlib /lib/ld-musl-mipsel-sf.so.1 --target mipsel-openwrt-linux-musl `cat /home/zxilly/lede/tmp/mklibs-progs /home/zxilly/lede/tmp/mklibs-libs` 2>&1
  File "/home/zxilly/lede/staging_dir/host/bin/mklibs", line 60
    print string.join(msg)
               ^
SyntaxError: invalid syntax
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/zxilly/lede'
package/Makefile:108: recipe for target '/home/zxilly/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/zxilly/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/zxilly/lede'
/home/zxilly/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

izhx, 2020-02-21:
错误信息差不多，win10，wsl ubuntu1604，最新的代码。
`
Enabling system
Enabling uhttpd
Enabling umount
./etc/init.d/unblockneteasemusic-mini: line 8: uci: command not found
./etc/init.d/unblockneteasemusic-mini: line 9: uci: command not found
Enabling unblockneteasemusic-mini
Enabling urandom_seed
Enabling urngd
Enabling v2ray_server
Enabling vsftpd
sed -i "s/Installed-Time: .*/Installed-Time: 1582284520/" /home/zhx/workspace/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/status
rm -rf /home/zhx/workspace/lede/build_dir/target-mipsel_24kc_musl/root-ramips/tmp/*
rm -f /home/zhx/workspace/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/lists/*
rm -f /home/zhx/workspace/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/info/*.postinst*
rm -f /home/zhx/workspace/lede/build_dir/target-mipsel_24kc_musl/root-ramips/var/lock/*.lock
rm -rf /home/zhx/workspace/lede/build_dir/target-mipsel_24kc_musl/root-ramips/boot
find /home/zhx/workspace/lede/build_dir/target-mipsel_24kc_musl/root-ramips/ -mindepth 1 -execdir touch -hcd "@1582284520" "{}" +
find: The relative path 'Files/WindowsApps/CanonicalGroupLimited.Ubuntu16.04onWindows_1604.2019.523.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/zhx/workspace/lede'
package/Makefile:108: recipe for target '/home/zhx/workspace/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/zhx/workspace/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/zhx/workspace/lede'
/home/zhx/workspace/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
`

Zxilly, 2020-02-21:
> 错误信息差不多，win10，wsl ubuntu1604，最新的代码。
>
> ```
> Enabling system
> Enabling uhttpd
> Enabling umount
> ./etc/init.d/unblockneteasemusic-mini: line 8: uci: command not found
> ./etc/init.d/unblockneteasemusic-mini: [...]

你是这个问题。。。
`find: The relative path 'Files/WindowsApps/CanonicalGroupLimited.Ubuntu16.04onWindows_1604.2019.523.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH`
不一样的。。。

azxcvjj, 2020-02-21:
我也报了，ubuntu14.04，百度一下怎么安装python3.5，装上就运行正常了。
现在大佬代码已经又更新了，支持老版本python

izhx, 2020-02-21:
确实，我关了WSL和windows的互操作和PATH导入，然后就好了。。。

0x5826, 2020-02-21:
按照以下两步操作：
1.在WSL的终端ubuntu中输入:
`echo "[interop]\nenabled=false\nappendWindowsPath=false" | sudo tee /etc/wsl.conf`
2.在powershell(以管理员身份运行)中输入: (以重启wsl)
```
net stop LxssManager
net start LxssManager
```

Ryuchen, 2020-02-21:
同样出现这个问题，WSL 关闭什么 就好了？

izhx, 2020-02-22:
> 同样出现这个问题，WSL 关闭什么 就好了？

https://blog.csdn.net/notmine/article/details/103302371

closed 2020-02-22 by coolsnowwolf

Windywind, 2020-03-08:
> 我也报了，ubuntu14.04，百度一下怎么安装python3.5，装上就运行正常了。
> 现在大佬代码已经又更新了，支持老版本python

为啥我不行呢？

```
/home/wind/lede/staging_dir/host/bin/mklibs -D -d /home/wind/lede/tmp/mklibs-out --sysroot /home/wind/lede/staging_dir/target-mips_24kc_musl/root-ar71xx `cat /home/wind/lede/tmp/mklibs-libs | sed 's:/*[^/]\+/*$::' | uniq | sed 's:^/home/wind/lede/staging_dir/target-mips_24kc_musl/root-ar71xx:-L :'` --ldlib /lib/ld-musl-mips-sf.so.1 --target mips-openwrt-linux-musl `cat /home/wind/lede/tmp/mklibs-progs /home/wind/lede/tmp/mklibs-libs` 2>&1
  File "/home/wind/lede/staging_dir/host/bin/mklibs", line 60
    print string.join(msg)
               ^
SyntaxError: invalid syntax
package/Makefile:65: recipe for target 'package/install' failed
```

fryderykhuang, 2020-03-18:
我也出一样的问题，说是print string.join() 是Python2的格式，但用的是python3所以出错。系统是ubuntu 18.04，python3.5额外装了，系统自带python3.6。

fryderykhuang, 2020-03-20:
> 为啥我不行呢？
>
> ```
> /home/wind/lede/staging_dir/host/bin/mklibs -D -d /home/wind/lede/tmp/mklibs-out --sysroot /home/wind/lede/staging_dir/target-mips_24kc_musl/root-ar71xx `cat [...]
问题解决了吗

1-1-2, 2020-05-28:
可能可行的临时解决方案
https://github.com/coolsnowwolf/lede/issues/3883#issuecomment-635419228