coolsnowwolf/lede #3252: 使用国外服务器编译报错 原因是国外vps访问mirrors.ustc.edu.cn 出错

luodaoyi, 2020-02-21:
估计是ip被国内墙掉了 所以访问不了 mirrors.ustc.edu.cn 然后就一直卡在这里

···
ubuntu@openwrt-build:~/lede$ make download V=s
make[1]: Entering directory '/home/ubuntu/lede'
make[2]: Entering directory '/home/ubuntu/lede/tools/gmp'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ubuntu/lede/tools/gmp'
time: tools/gmp/download#0.02#0.00#0.03
make[2]: Entering directory '/home/ubuntu/lede/tools/mpfr'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/ubuntu/lede/tools/mpfr'
time: tools/mpfr/download#0.02#0.00#0.03
make[2]: Entering directory '/home/ubuntu/lede/tools/mpc'
mkdir -p /home/ubuntu/lede/dl
SHELL= flock /home/ubuntu/lede/tmp/.mpc-1.1.0.tar.gz.flock -c '         /home/ubuntu/lede/scripts/download.pl "/home/ubuntu/lede/dl" "mpc-1.1.0.tar.gz" "6985c538143c1208dcb1ac42cedad6ff52e267b47e5f970183a3e75125b43c2e" "" "@GNU/mpc/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirrors.ustc.edu.cn/gnu/mpc/mpc-1.1.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
···

luodaoyi, 2020-02-21:
又啥办法可以关掉这个 mrrors 直接用老外的呢

luodaoyi, 2020-02-21:
```
把目录：
scripts/download.pl
中ustc相关的mirrors 注释掉好像可以了 

```

AmadeusGhost, 2020-02-22:
这个地址用不了会自动使用下一个地址，而且utsc这个下载地址以前就加进去了

luodaoyi, 2020-02-22:
> 这个地址用不了会自动使用下一个地址，而且utsc这个下载地址以前就加进去了

对的 我也发现了

closed 2020-02-22 by luodaoyi