coolsnowwolf/lede #3254: 貌似是v2ray编译不过吗?

git-maven, 2020-02-21:
/home/bibi/work/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src does not exist, skipping symlinks

touch /home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.configured_1cb8d937842536ed4b0e7e1ea908129b
rm -f /home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.built
touch /home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.built_check
( cd /home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/build ; export GOPATH=/home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/build GOCACHE=/home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/cache GOTMPDIR=/home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/tmp GOROOT_FINAL=/usr/lib/go-1.13 CC=x86_64-openwrt-linux-musl-gcc CXX=x86_64-openwrt-linux-musl-g++ 	GOOS=linux GOARCH=amd64 GO386=387 GOARM= GOMIPS= GOMIPS64= CGO_ENABLED=1 CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1:v2ray-core-4.22.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/bibi/work/lede/staging_dir/target-x86_64_musl/usr/include -I/home/bibi/work/lede/staging_dir/target-x86_64_musl/include -I/home/bibi/work/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/bibi/work/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/bibi/work/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1:v2ray-core-4.22.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" ; echo "Finding targets" ; targets=$(go list v2ray.com/core/main) ; for pattern in ; do targets=$(echo "$targets" | grep -v "$pattern") ; done ; echo ; if [ "" = 1 ]; then echo "Calling go generate" ; go generate -v  $targets ; echo ; fi ; if [ "" != 1 ]; then echo "Building targets" ; case amd64 in arm)             installsuffix="v" ;; mips|mipsle)     installsuffix="" ;; mips64|mips64le) installsuffix="" ;; esac ; ldflags="-linkmode external -extldflags '-L/home/bibi/work/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/bibi/work/lede/staging_dir/target-x86_64_musl/lib -L/home/bibi/work/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/lib -L/home/bibi/work/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib -Wl,-z,now -Wl,-z,relro'" ; pkg_gcflags="" ; pkg_ldflags="-s -w" ; for def in v2ray.com/core.version=4.22.1 v2ray.com/core.build=Lean v2ray.com/core.codename=OpenWrt; do pkg_ldflags="$pkg_ldflags -X $def" ; done ; go install ${installsuffix:+-installsuffix $installsuffix} -trimpath -ldflags "all=$ldflags" -v ${pkg_gcflags:+-gcflags "$pkg_gcflags"} ${pkg_ldflags:+-ldflags "$pkg_ldflags $ldflags"}  $targets ; retval=$? ; echo ; if [ "$retval" -eq 0 ] && [ -z "$(find /home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.go_work/build/bin -maxdepth 0 -type d \! -empty 2>/dev/null)" ]; then echo "No binaries were generated, consider adding GO_PKG_SOURCE_ONLY:=1 to Makefile" ; echo ; fi ; echo "Cleaning module download cache (golang/go#27455)" ; go clean -modcache ; echo ; fi ; exit $retval ; )
Finding targets
go: github.com/golang/mock@v1.2.0: Get https://proxy.golang.org/github.com/golang/mock/@v/v1.2.0.mod: proxyconnect tcp: EOF

Building targets
go: github.com/golang/mock@v1.2.0: Get https://proxy.golang.org/github.com/golang/mock/@v/v1.2.0.mod: proxyconnect tcp: EOF

Cleaning module download cache (golang/go#27455)

Makefile:307: recipe for target '/home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.built' failed
make[3]: *** [/home/bibi/work/lede/build_dir/target-x86_64_musl/v2ray-core-4.22.1/.built] Error 1
make[3]: Leaving directory '/home/bibi/work/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#1.81#1.03#2.80
package/Makefile:111: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/bibi/work/lede'
package/Makefile:107: recipe for target '/home/bibi/work/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/bibi/work/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bibi/work/lede'
/home/bibi/work/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

重新clone了几次都是报同样的错,不知各位有没有这个问题?

SharerMax, 2020-02-21:
确保使用正确的网络环境，依赖没拉下来

feizaibufei, 2020-02-22:
不知道为什么我编译N1固件V2RAY启动不能上网?

AmadeusGhost, 2020-02-22:
go可以使用国内代理
```
export GO111MODULE=on
export GOPROXY=https://goproxy.cn
```

git-maven, 2020-02-22:
> go可以使用国内代理
> 
> ```
> export GO111MODULE=on
> export GOPROXY=https://goproxy.cn
> ```

请这两行代码添加在哪里呢?直接在控制台输入就行了吗?

Mattraks, 2020-02-22:
国内环境启用V2RAY_COMPRESS_GOPROXY

feizaibufei, 2020-02-22:
问题是我编译好的启动V2RAY 国内 国外 都不能上

feizaibufei, 2020-02-22:
十六号编译的固件上网正常。

git-maven, 2020-02-22:
添加go国内代理以后出现404..

Finding targets
go: github.com/v2ray/v2ray-core@v4.22.1+incompatible: reading https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod: 404 Not Found

Building targets
go: github.com/v2ray/v2ray-core@v4.22.1+incompatible: reading https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod: 404 Not Found

Cleaning module download cache (golang/go#27455)

Makefile:59: recipe for target '/home/bibi/work/lean-lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0/.built' failed
make[3]: *** [/home/bibi/work/lean-lede/build_dir/target-x86_64_musl/v2ray-plugin-1.3.0/.built] Error 1
make[3]: Leaving directory '/home/bibi/work/lean-lede/package/lean/v2ray-plugin'
time: package/lean/v2ray-plugin/compile#0.19#0.17#1.78
package/Makefile:111: recipe for target 'package/lean/v2ray-plugin/compile' failed
make[2]: *** [package/lean/v2ray-plugin/compile] Error 2
make[2]: Leaving directory '/home/bibi/work/lean-lede'
package/Makefile:107: recipe for target '/home/bibi/work/lean-lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/bibi/work/lean-lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bibi/work/lean-lede'
/home/bibi/work/lean-lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

SharerMax, 2020-02-22:
> https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod

最好还是保证科学上网环境后再来编译，或者如果你没有 v2ray-plugin的需求，你可以不勾选

git-maven, 2020-02-22:
> > https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod
> 
> 最好还是保证科学上网环境后再来编译，或者如果你没有 v2ray-plugin的需求，你可以不勾选

一直科学上网来编译,你可以进去这个网址看看,返回也是not found

git-maven, 2020-02-22:
> > https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod
> 
> 最好还是保证科学上网环境后再来编译，或者如果你没有 v2ray-plugin的需求，你可以不勾选

好像说不允许使用+incompatible.mod

SharerMax, 2020-02-22:
> > > https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod
> > 
> > 
> > 最好还是保证科学上网环境后再来编译，或者如果你没有 v2ray-plugin 的需求，你可以不勾选
> 
> 好像说不允许使用 + incompatible.mod

不要用go的镜像代理服务，保证科学上网环境，就对了

git-maven, 2020-02-22:
查了下资料是go版本为1.3的不允许使用+incompatible.mod
正在尝试把系统的go环境降级到1.12

naoki66, 2020-02-26:
同样问题

SharerMax, 2020-02-27:
编译v2ray-plugin请不要使用任何go的镜像代理，全局科学上网就好
PS: v2ray和v2ray-plugin是两个东西，如果你只是想用v2ray，就不必勾选v2ray-plugin

lizhenglin2002, 2020-02-28:
我也是这个问题，请问怎样解决

closed 2020-03-06 by coolsnowwolf

Ghost-Assassin, 2020-03-08:
> 我也是这个问题，请问怎样解决
我的的解决方法是在编译前先执行
`export GOPROXY=https://goproxy.cn,direct`
然后不需要挂全局代理也可以正常编译