coolsnowwolf/lede #3259: 报告，我似乎发现一个错误。

vodott, 2020-02-22:
我编译了3次都是出现这个错误。第一次是在二次编译的环境下，第二次使用了 make clean之后，第三次是在make distclean之后。我是新手，请大家不吝赐教，感恩！

make[6]: Leaving directory `/home/zhangqi/lede/build_dir/hostpkg/libtasn1-4.16.0/src'
Making all in fuzz
make[6]: Entering directory `/home/zhangqi/lede/build_dir/hostpkg/libtasn1-4.16.0/fuzz'
gcc -DHAVE_CONFIG_H -I. -I..  -I../lib/includes -I../lib/includes -I. -I../lib/gl -I../lib/gl -DSRCDIR=\"/home/zhangqi/lede/build_dir/hostpkg/libtasn1-4.16.0/fuzz\"  -DTEST_RUN -I/home/zhangqi/lede/staging_dir/host/include -I/home/zhangqi/lede/staging_dir/hostpkg/include -I/home/zhangqi/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include  -fno-common -W -Waddress -Waggressive-loop-optimizations -Wall -Wattributes -Wbad-function-cast -Wbuiltin-macro-redefined -Wcast-align -Wchar-subscripts -Wclobbered -Wcomment -Wcomments -Wcoverage-mismatch -Wcpp -Wdeprecated -Wdeprecated-declarations -Wdisabled-optimization -Wdiv-by-zero -Wdouble-promotion -Wempty-body -Wendif-labels -Wenum-compare -Wextra -0Wformat-contains-nul -Wformat-extra-args -Wformat-nonliteral -Wformat-security -Wformat-y2k -Wformat-zero-length -Wfree-nonheap-object -Wignored-qualifiers -Wimplicit -Wimplicit-function-declaration -Wimplicit-int -Winit-self -Winline -Wint-to-pointer-cast -Winvalid-memory-model -Winvalid-pch -Wlogical-op -Wmain -Wmaybe-uninitialized -Wmissing-braces -Wmissing-declarations -Wmissing-field-initializers -Wmissing-include-dirs -Wmissing-parameter-type -Wmissing-prototypes -Wmultichar -Wnarrowing -Wnested-externs -Wnonnull -Wold-style-declaration -Wold-style-definition -Woverflow -Woverride-init -Wpacked -Wpacked-bitfield-compat -Wparentheses -Wpointer-arith -Wpointer-sign -Wpointer-to-int-cast -Wpragmas -Wpsabi -Wreturn-local-addr -Wreturn-type -Wsequence-point -Wshadow -Wsizeof-pointer-memaccess -Wstack-protector -Wstrict-aliasing -Wstrict-overflow -Wstrict-prototypes -Wsuggest-attribute=const -Wsuggest-attribute=format -Wsuggest-attribute=noreturn -Wswitch -Wsync-nand -Wtrampolines -Wtrigraphs -Wtype-limits -Wuninitialized -Wunknown-pragmas -Wunsafe-loop-optimizations -Wunused -Wunused-but-set-parameter -Wunused-but-set-variable -Wunused-function -Wunused-label -Wunused-local-typedefs -Wunused-parameter -Wunused-result -Wunused-value -Wunused-variable -Wvarargs -Wvariadic-macros -Wvector-operation-performance -Wvla -Wvolatile-register-var -Wwrite-strings -Wnormalized=nfc -Wno-type-limits -Wno-unused-parameter -fdiagnostics-show-option  -O2 -I/home/zhangqi/lede/staging_dir/host/include -I/home/zhangqi/lede/staging_dir/hostpkg/include -I/home/zhangqi/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include -MT corpus2array.o -MD -MP -MF .deps/corpus2array.Tpo -c -o corpus2array.o corpus2array.c
cc1: warning: /home/zhangqi/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include: No such file or directory [enabled by default]
cc1: warning: /home/zhangqi/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/host/include: No such file or directory [enabled by default]
corpus2array.c: In function 'main':
corpus2array.c:123:4: error: 'for' loop initial declarations are only allowed in C99 mode
    for (int i = 8; i < 32 ; i++) {
    ^
corpus2array.c:123:4: note: use option -std=c99 or -std=gnu99 to compile your code
make[6]: *** [corpus2array.o] Error 1
make[6]: Leaving directory `/home/zhangqi/lede/build_dir/hostpkg/libtasn1-4.16.0/fuzz'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/zhangqi/lede/build_dir/hostpkg/libtasn1-4.16.0'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/zhangqi/lede/build_dir/hostpkg/libtasn1-4.16.0'
make[3]: *** [/home/zhangqi/lede/build_dir/hostpkg/libtasn1-4.16.0/.built] Error 2
make[3]: Leaving directory `/home/zhangqi/lede/feeds/packages/libs/libtasn1'
time: package/feeds/packages/libtasn1/host-compile#5.67#0.95#9.76
make[2]: *** [package/feeds/packages/libtasn1/host/compile] Error 2
make[2]: Leaving directory `/home/zhangqi/lede'
make[1]: *** [/home/zhangqi/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/zhangqi/lede'
make: *** [world] 错误 2

closed 2020-02-22 by coolsnowwolf

AmadeusGhost, 2020-02-22:
这是给什么设备编译的？用什么系统编译？

vodott, 2020-02-22:
设备是Linksys001 wrt3200acm，编译环境是Ubuntu 14.04.6。在20200216和20200219我都成功编译过，但昨晚我在Global build setting中取消了“enable IPv6 support in packages”。难道是因为取消这个选项？

vodott, 2020-02-22:
> 这是给什么设备编译的？用什么系统编译？

设备是Linksys001 wrt3200acm，编译环境是Ubuntu 14.04.6。在20200216和20200219我都成功编译过，但昨晚我在Global build setting中取消了“enable IPv6 support in packages”。难道是因为取消这个选项？

AmadeusGhost, 2020-02-22:
建议用Ubuntu 18.04编译，还有有些选项最好乱动