coolsnowwolf/lede #3261: DNSPOD未能正常运行

wspzf, 2020-02-22:
硬件为软路由
一个月前编译的DNSPOD能正常运行
最近几次编译的DNSPOD都未能正常运行，无dnspod进程，/tmp目录无dnspod.log日志文件
手动运行会报错，
root@OpenWrt:~# ps -ef|grep dnspod
 8691 root      1072 S    grep dnspod
root@OpenWrt:/# ./etc/init.d/dnspod start
/etc/rc.common: root@OpenWrt:/# line 6: /usr/sbin/dnspod.sh: not found

下面是一个月前编译的固件，开机后有dnspod进程
root@OpenWrt:~# ps -ef|grep dnspod
 4077 root      2204 S    {dnspod.sh} /bin/bash /usr/sbin/dnspod.sh --svc
手动执行./etc/init.d/dnspod start后无报错，会多一个dnspod进程

wspzf, 2020-02-22:
> 硬件为软路由
> 一个月前编译的DNSPOD能正常运行
> 最近几次编译的DNSPOD都未能正常运行，无dnspod进程，/tmp目录无dnspod.log日志文件
> 手动运行会报错，
> root@OpenWrt:~# ps -ef|grep dnspod
> 8691 root 1072 S grep dnspod
> root@OpenWrt:/# ./etc/init.d/dnspod start [...]

最近因为编译报错，执行过 make clean，难道有什么依赖没有选择上？

hcym, 2020-02-22:
大雕的就是这个头疼，

coolsnowwolf, 2020-02-22:
./etc/init.d/dnspod 早已删除，现在是内置ddns 的版本，你在哪里搞的？

coolsnowwolf, 2020-02-22:
> 大雕的就是这个头疼，

就是哪个头疼？阴阳怪气的

hcym, 2020-02-22:
哦，没有啦，就是dns转发有时会无症状失效，怎么搞都不行，只能重启看看，有时又非常稳定，怎么切换都没事。头疼，不过出差的机会很少。

closed 2020-02-22 by coolsnowwolf