coolsnowwolf/lede #3262: 依赖缺失：libncursesw6.1

SharerMax, 2020-02-22:
代码版本：
>https://github.com/coolsnowwolf/lede/tree/62ec3bfa9de27bffd60de389f8f445b2807021b3

编译环境：
> ubuntu-18.04 server x64

期望：
>编译通过

问题：
>勾选  luci-app-haproxy-tcp 后编译报错，检查发现 haproxy 依赖libncursesw，但是我并没有找到这个库

已尝试的解决方法：
> 更新feeds 后 删除 .config 和 ./tmp 后重新编译

日志如下：

```
Configuring bcm27xx-gpu-fw.
Configuring urngd.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-haproxy-tcp:
 * 	libncursesw6.1
 * opkg_install_cmd: Cannot install package luci-app-haproxy-tcp.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/max/code/lean-lede'
package/Makefile:108: recipe for target '/home/max/code/lean-lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_install' failed
make[1]: *** [/home/max/code/lean-lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/max/code/lean-lede'
/home/max/code/lean-lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

coolsnowwolf, 2020-02-22:
make clean 这个 luci 包，因为 libncursesw6.1 官方改名为 libncursesw6 了

coolsnowwolf, 2020-02-22:
libncursesw 是 libncurses 的别名

closed 2020-02-22 by coolsnowwolf