coolsnowwolf/lede #3264: 编译出错

qdfansanyblog, 2020-02-22:
Configuring luci-app-ramfree.
Configuring luci-i18n-ramfree-zh-cn.
Configuring luci-app-wol.
Configuring kmod-usb-net-rtl8150.
Configuring kmod-usb-net-rtl8152.
Configuring kmod-tulip.
Configuring luci-i18n-wol-zh-cn.
Configuring luci-app-upnp.
Configuring ath10k-firmware-qca9888.
Configuring luci-proto-relay.
Configuring default-settings.
Configuring htop.
Configuring luci-theme-netgear.
Configuring kmod-i2c-algo-bit.
Configuring kmod-igbvf.
Configuring luci-i18n-base-zh-cn.
Configuring ath10k-firmware-qca9984.
Configuring ath10k-firmware-qca988x.
Configuring fdisk.
Configuring autosamba.
Configuring kmod-e1000e.
Configuring kmod-ath9k-htc.
Configuring luci-i18n-upnp-zh-cn.
Configuring libustream-openssl20150806.
Configuring kmod-button-hotplug.
Configuring kmod-igb.
Configuring kmod-bnx2.
Configuring urngd.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for alsa-utils:
 *      libncursesw6.1
 * opkg_install_cmd: Cannot install package alsa-utils.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/wwx/lede'
package/Makefile:108: recipe for target '/home/wwx/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/wwx/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/wwx/lede'
/home/wwx/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

EnnawYang, 2020-02-22:
https://github.com/coolsnowwolf/lede/issues/3262

EstFoxs, 2020-02-22:
alsa-utils删除这个包，重新编译

closed 2020-02-22 by coolsnowwolf