coolsnowwolf/lede #3266: 有没有人知道这是怎么回事啊？？

ppp47, 2020-02-22:
https://whoer.net/zh#
同一个节点在pc上使用代理软件与在op路由器上使用，怎么会出现两种检测情况？？？
还是说我的路由器设置的有问题。

在pc使用
[image]
[image]

在op路由器上使用
[image]
[image]

monsterzzzz, 2020-02-22:
感觉是DNS解析问题，如果全通过代理解析IP，就不会出现这个

ppp47, 2020-02-22:
> 感觉是DNS解析问题，如果全通过代理解析IP，就不会出现这个

我不懂，请问具体该怎样操作呢？

9l, 2020-02-23:
你直接用路由器上网的么？
电脑上dns没问题就行了，别自寻烦恼

ppp47, 2020-02-23:
> 你直接用路由器上网的么？
> 电脑上dns没问题就行了，别自寻烦恼

我是把op路由器连上电脑进行上网的

ppp47, 2020-02-23:
> https://whoer.net/zh#
> 同一个节点在pc上使用代理软件与在op路由器上使用，都是全局代理,怎么会出现两种DNS检测结果？？？
> 还是说我的路由器设置的有问题。
> 
> 在pc使用
> [image]
> [image]
> 
> 在op路由器上使用
> [image]
> [image]

9l, 2020-02-23:
你路由开全局模式试试

ppp47, 2020-02-23:
> 你路由开全局模式试试

是开全局模式的
[image]

closed 2020-03-06 by coolsnowwolf