coolsnowwolf/lede #3268: Samba4-libs云编译出错，缺少依赖libpam（Makefile中声明了该依赖）

hue715, 2020-02-22:
**报错代码：**
cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/samba-4.11.6/ipkg-install/usr/lib/samba /home/runner/work/lede/lede/build_dir/target-x86_64_musl/samba-4.11.6/ipkg-x86_64/samba4-libs/usr/lib/
2020-02-22T07:20:41.1487304Z find /home/runner/work/lede/lede/build_dir/target-x86_64_musl/samba-4.11.6/ipkg-x86_64/samba4-libs -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package samba4-libs is missing dependencies for the following libraries:libpam.so.0
Makefile:430: recipe for target '/home/runner/work/lede/lede/bin/packages/x86_64/base/samba4-libs_4.11.6-1_x86_64.ipk' failed
**libpam未被强制勾选，如下图**
[image]
**在samba4的makefile中，samba-libs的依赖包含libpam**
[image]

bluehj777, 2020-02-22:
同一问题
如果要添加编译luci-app-samba4，需要手动把这个libpam手动加上，否则无法编译成功，加上后则顺利编译成功

Aes64X, 2020-02-22:
> 同一问题
> 如果要添加编译luci-app-samba4，需要手动把这个libpam手动加上，否则无法编译成功，加上后则顺利编译成功

话说。。。。samba4我基本都不看。。。。你们是拿来看视频吗。。。。 我都有NFS - -#

coolsnowwolf, 2020-02-22:
rm -rf tmp 就会关联上

hue715, 2020-02-22:
> rm -rf tmp 就会关联上

试了一下，清除tmp，再make menuconfig并选择samba4-libs，仍然没有关联。

hyiqun, 2020-02-28:
我是新手，一直编译失败，绕了很多弯路，添加依赖后问题解决，感谢。

hue715, 2020-03-02:
#

closed 2020-03-02 by hue715