coolsnowwolf/lede #3272: linksys1900系列无线延迟高

segadrive, 2020-02-22:
关闭wmm延迟能降低二十几ms，是不是驱动的问题

Aes64X, 2020-02-22:
> 关闭wmm延迟能降低二十几ms，是不是驱动的问题

Linksys WRT系列和几个老型号使用 Marvell 方案，此方案源码开源并由 Linksys 官方维护，请到官方支持站点看看。

segadrive, 2020-02-22:
> > 关闭wmm延迟能降低二十几ms，是不是驱动的问题
> 
> Linksys WRT系列和几个老型号使用 Marvell 方案，此方案源码开源并由 Linksys 官方维护，请到官方支持站点看看。

好的收到

rageq, 2020-02-28:
WRT系列无线延迟一直很高，高通的能15ms左右，他能去到30ms左右
放弃吧，高通AP加软路由是出路

segadrive, 2020-02-28:
> WRT系列无线延迟一直很高，高通的能15ms左右，他能去到30ms左右
> 放弃吧，高通AP加软路由是出路

你好，我用openwrt测试延迟就是15ms，而且也找到了最新的驱动，请问怎么将驱动编译进去，https://github.com/eduperez/mwlwifi_LEDE

rageq, 2020-02-28:
> 
> 
> > WRT系列无线延迟一直很高，高通的能15ms左右，他能去到30ms左右
> > 放弃吧，高通AP加软路由是出路
> 
> 你好，我用openwrt原版测试延迟就是15ms，而且也找到了最新的驱动，请问怎么将驱动编译进去，https://github.com/eduperez/mwlwifi_LEDE

替换\package\kernel\mwlwifi里面的文件
或者修改里面的版本号也行
https://github.com/kaloz/mwlwifi
这个才是半官方维护的代码

segadrive, 2020-02-28:
> > > WRT系列无线延迟一直很高，高通的能15ms左右，他能去到30ms左右
> > > 放弃吧，高通AP加软路由是出路
> > 
> > 
> > 你好，我用openwrt原版测试延迟就是15ms，而且也找到了最新的驱动，请问怎么将驱动编译进去，https://github.com/eduperez/mwlwifi_LEDE
> 
> 替换\package\kernel\mwlwifi里面的文件
> 或者修改里面的版本号也行
> https://github.com/kaloz/mwlwifi
> 这个才是半官方维护的代码

奇怪，我替换了还是不行，莫非官方的固件有神秘加持?

segadrive, 2020-02-29:
> > > WRT系列无线延迟一直很高，高通的能15ms左右，他能去到30ms左右
> > > 放弃吧，高通AP加软路由是出路
> > 
> > 
> > 你好，我用openwrt原版测试延迟就是15ms，而且也找到了最新的驱动，请问怎么将驱动编译进去，https://github.com/eduperez/mwlwifi_LEDE
> 
> 替换\package\kernel\mwlwifi里面的文件
> 或者修改里面的版本号也行
> https://github.com/kaloz/mwlwifi
> 这个才是半官方维护的代码

大佬，我百科到问题了，说是禁用了amsdu就行了，但是在哪儿禁用呢，我编译进去就一劳永逸了，感激不尽

closed 2020-03-06 by coolsnowwolf