coolsnowwolf/lede #3275: 关于 IPv6 和 DNS 缓存的 DNSMASQ 设置说明

coolsnowwolf, 2020-02-22:
新版 DNSMASQ 我打了两个关键的补丁：

1. IPv6 AAAA 记录过滤补丁。众所周知国内 IPv6 其实根本没有成熟，开了 IPv6 对日常访问网站和科学上网是一种副作用。

所以新固件修改了 DNSMASQ ，默认对 DNS 查询只返回 IPv4 的 A 记录，丢弃同一域名的 IPv6 AAAA 记录。

这样的好处就是内网的 P2P 等下载依然可以在 IPv6 中获益，而日常上网又不会受到系统默认 IPv6 优先的干扰

[image]

2. 客户端最小 MINI TTL 修改功能。 这个功能是第一次返回 DNS 记录的时候，把 DNS 的 TTL 最小时间修改，例如可以改为 3600 秒 一小时。 这样有利于把域名查询结果更长时间缓存在客户端内存中，而不必再次发起 DNS 查询，0延迟

[image]

大家对这两个功能有何建议的，欢迎提出

klever1988, 2020-02-22:
屏蔽aaaa记录可能会造成谷歌应用无法联网，之前我在走ssrp的pdnsd屏蔽aaaa记录遇到这个问题

hxlls, 2020-02-22:
dns 缓存这个加速功能缓存的是不是也可以生效TLL，如果可以，建议就是把dnsmasq 的对应设置移动到DNS加速那去设置，因为目前的设置比较混乱，dns 加速功能只有填写一堆DNS IP地址，但是并发查询又再DNS设置那勾选，还有TTL数值也在DNS那选择，如果做到一个页面上，应该会更方便设置。还有一个，就是如果可能，能支持DoH 查询,然后缓存到本机最好，HTTPS 理论上劫持是最少的

madcat0513, 2020-02-23:
可能还有国外需要ipv6的，做两个选项就可以啦，有这个需要的自己开启就行了。

you-mu, 2020-02-24:
IPv6 AAAA 记录过滤 蛮有用的,
挂PT要用到ipv6,又想兼顾科学上网.

coolsnowwolf, 2020-02-25:
并不会影响 DDNS 更新，你将 IP 来源设置为网络 WAN6 即可

[image]

you-mu, 2020-02-25:
> 
> 
> 并不会影响 DDNS 更新，你将 IP 来源设置为网络 WAN6 即可
> 
> [image]

好嘞,感谢L大.

不过我的DDNS不是在路由器上,是在我的 NAS上.

[image]

hcym, 2020-02-25:
使用ipv6没有影响，就是状态栏返回失败，但是有ping值，ping百度6失败ping谷歌6正常。😄

31551, 2020-02-25:
IPV6还是很重要的。目前国外的VPS服务器IPV4地址被墙了，IPV6照常解析，速度还很快。

hcym, 2020-02-25:
我的移动去年不错，有时比ipv4的快，而且能跑满宽带，今年惨了，和套ws的差不多，兵荒马乱时求一个心安吧😂

wybb, 2020-02-25:
这个想法在去年年底我也考虑到了，不过我觉得你处理还可以更加优化一些，DNSMASQ 在同时存在ipv4和ipv6地址的情况下，去除ipv6的地址没问题，但是当上游查询结果仅有AAAA记录的情况下应该不进行过滤，这样可以保证日常都是优先ip4，但是纯ipv6的域名也应该能够正确访问

Her0R0cket, 2020-02-26:
昨天捣鼓的时候也发现了这个功能,已经用上了,非常感谢,还以为这个官方功能来着

makepoloSeton, 2020-02-27:
很不错的补丁，许多APP支持ipv6后打开页面的速度慢了不少（京东，淘宝等），但是对于没有ipv4公网地址的人来说，ipv6的公网可以用来远程访问，p2p加速，解决这个问题的一个方法就是屏蔽掉ipv6的解析结果，这样既能带来公网远程访问家里，在下载时连接更多的节点等好处，又能屏蔽掉国内ipv6不成熟却硬上马带来的访问延迟问题，有特殊访问需求的ipv6网站可以添加hosts来解决，这确实是一个不错的功能。

aepub, 2020-02-29:
WAN6端口起什么作用，我以及把它给删除了！需要增加回来吗，怎样增加！

Sevenzhi, 2020-03-05:
所以小白想问一下禁止解析 IPv6 DNS 记录到底要不要打上√，小白看不懂，勾上了IPv6 连接测试会是0分

closed 2020-03-06 by coolsnowwolf

klever1988, 2020-03-29:
#https://github.com/coolsnowwolf/lede/issues/4087

DragonCat1, 2021-01-27:
@coolsnowwolf ipv6的过滤有没有可能指定ip段，比如只过滤国外不过滤国内

kongfl888, 2022-09-28:
回来告知
dnsmasq 2.87（2天前）
已经加入了
--filter-AAAA
现在等op做出更新了

kongfl888, 2022-09-28:
https://github.com/openwrt/openwrt/pull/10820