coolsnowwolf/lede #3276: 更新源码后编译出错

tcsr200722, 2020-02-22:
* satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-samba4:
 * 	samba4-libs
 * opkg_install_cmd: Cannot install package luci-app-samba4.
 * check_conflicts_for: The following packages conflict with samba4-server:
 * check_conflicts_for: 	samba36-server * 
 * opkg_install_cmd: Cannot install package samba4-server.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for alsa-utils:
 * 	libncursesw6.1
 * opkg_install_cmd: Cannot install package alsa-utils.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/lenovo/OpenWrt'
package/Makefile:108: recipe for target '/home/lenovo/OpenWrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/lenovo/OpenWrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/lenovo/OpenWrt'
/home/lenovo/OpenWrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

505799690, 2020-02-22:
一样的问题
install -m0755 /workspace/lede/build_dir/hostpkg/samba-4.11.6/bin/compile_et /workspace/lede/staging_dir/hostpkg/bin/compile_et_samba
mkdir -p /workspace/lede/staging_dir/hostpkg/stamp
touch /workspace/lede/build_dir/hostpkg/samba-4.11.6/.built
touch /workspace/lede/staging_dir/hostpkg/stamp/.samba_installed
make[3]: Leaving directory '/workspace/lede/package/lean/samba4'
time: package/lean/samba4/host-compile#102.71#70.06#215.54
make[2]: Leaving directory '/workspace/lede'
make[1]: *** [package/Makefile:107: /workspace/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workspace/lede'
make: *** [/workspace/lede/include/toplevel.mk:227: world] Error 2

lloyd18, 2020-02-22:
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for alsa-utils:
 * 	libncursesw6.1
 * opkg_install_cmd: Cannot install package alsa-utils.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/home/lan/lede'
make[1]: *** [package/Makefile:108: /home/lan/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/lan/lede'
make: *** [/home/lan/lede/include/toplevel.mk:227：world] 错误 2

lqhorochi, 2020-02-22:
一样的问题+1

ckxt, 2020-02-22:
[lede.txt](https://github.com/coolsnowwolf/lede/files/4239701/lede.txt)

我碰到了一样的问题，附上单线程编译的logs，看了半天没看出来哪里出问题了，哪位有空的话帮我看下，谢谢

retusus, 2020-02-22:
> [lede.txt](https://github.com/coolsnowwolf/lede/files/4239701/lede.txt)
> 
> 我碰到了一样的问题，附上单线程编译的logs，看了半天没看出来哪里出问题了，哪位有空的话帮我看下，谢谢

ntfs-3g和antfs-mount冲突，取消勾选antfs-mount

retusus, 2020-02-22:
> * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-samba4:
> * samba4-libs
> * opkg_install_cmd: Cannot install package luci-app-samba4.
> * check_conflicts_for: The [...]
samba36-server和samba4-server冲突，

AmadeusGhost, 2020-02-23:
`make clean`后重新编译即可

ckxt, 2020-02-23:
> `make clean`后重新编译即可

谢谢！ 真的解决了！！

ckxt, 2020-02-23:
> > [lede.txt](https://github.com/coolsnowwolf/lede/files/4239701/lede.txt)
> > 我碰到了一样的问题，附上单线程编译的logs，看了半天没看出来哪里出问题了，哪位有空的话帮我看下，谢谢
> 
> ntfs-3g和antfs-mount冲突，取消勾选antfs-mount

您好，谢谢你的回复，我用了楼下的 make clean命令也解决了，感谢！

you-mu, 2020-02-24:
编译遇到问题,可以先在issue里面搜索.
而且其实make clean 可以解决很多问题了.

PR #3496 (2020-03-03): CI: add g++-multilib to dependence
cpufreq这个感谢指正，但是我不会写2333
就不重新开一个pr了，close #3510 
顺便 close #3270 , close #3276 , close #3281 , close #3291 , close #3336 , close #3414 .
closed 2020-03-03 by coolsnowwolf