coolsnowwolf/lede #3278: 编译一个京东云无线宝固件，但是没有自动挂载内置的100Gssd，为什么？

shenzt68, 2020-02-22:
如题
[image]
但是恩山这个帖子https://www.right.com.cn/forum/thread-2846333-1-1.html的固件是有100G的

ricksuzade-maker, 2020-02-22:
没有格式化？

AmadeusGhost, 2020-02-23:
那是emmc，，，
自己确认下`kmod-mmc` `kmod-sdhci` `kmod-mmc-spi`有没有选上

sky123456789sky, 2020-02-24:
最好确定下ext4有没有装 我编译极路由的时候咋都挂载不上 然后按网上教程各种驱动一动装 然后最后发现啥都不用装 装个EXT4就行了

shenzt68, 2020-02-25:
已经装上了EXT4
[image]

shenzt68, 2020-02-25:
用fdisk查看，好像分区表错误！
root@OpenWrt:~# fdisk -lu
Disk /dev/mmcblk0: 115 GB, 123731968000 bytes, 241664000 sectors
3776000 cylinders, 4 heads, 16 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mmcblk0 doesn't contain a valid partition table
Disk /dev/mmcblk0boot1: 4 MB, 4194304 bytes, 8192 sectors
128 cylinders, 4 heads, 16 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mmcblk0boot1 doesn't contain a valid partition table
Disk /dev/mmcblk0boot0: 4 MB, 4194304 bytes, 8192 sectors
128 cylinders, 4 heads, 16 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mmcblk0boot0 doesn't contain a valid partition table

closed 2020-03-06 by coolsnowwolf

zhanglei2021, 2021-09-29:
可否提供一下京东云的编译方法和配置文件

dudq786, 2024-04-11:
这种该怎么操作啊   跟教程完全不一样啊

[JDC-1 /opt/home/admin]# fdisk -lu
Disk /dev/mtdblock0: 0 MB, 196608 bytes, 384 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock0 doesn't contain a valid partition table
Disk /dev/mtdblock1: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock1 doesn't contain a valid partition table
Disk /dev/mtdblock2: 0 MB, 65536 bytes, 128 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock2 doesn't contain a valid partition table
Disk /dev/mtdblock3: 1 MB, 1395200 bytes, 2725 sectors
0 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock3 doesn't contain a valid partition table
Disk /dev/mtdblock4: 19 MB, 20362240 bytes, 39770 sectors
2 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock4 doesn't contain a valid partition table
Disk /dev/mtdblock5: 10 MB, 11468800 bytes, 22400 sectors
1 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock5 doesn't contain a valid partition table
Disk /dev/mtdblock6: 20 MB, 21757952 bytes, 42496 sectors
2 cylinders, 255 heads, 63 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mtdblock6 doesn't contain a valid partition table
Found valid GPT with protective MBR; using GPT

Disk /dev/mmcblk0: 241664000 sectors, 3312M
Logical sector size: 512
Disk identifier (GUID): f59a318b-40d2-2f4a-82d2-9e3b1162c2ff
Partition table holds up to 128 entries
First usable sector is 2048, last usable sector is 241663966

Number  Start (sector)    End (sector)  Size Name
     1            2048          206847  100M
     2          206848          411647  100M
     3          411648         2508799 1024M
     4         2508800       238518271  112G
     5       238518272       239566847  512M
     6       239566848       241663966 1023M
Disk /dev/mmcblk0boot1: 4 MB, 4194304 bytes, 8192 sectors
128 cylinders, 4 heads, 16 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mmcblk0boot1 doesn't contain a valid partition table
Disk /dev/mmcblk0boot0: 4 MB, 4194304 bytes, 8192 sectors
128 cylinders, 4 heads, 16 sectors/track
Units: sectors of 1 * 512 = 512 bytes

Disk /dev/mmcblk0boot0 doesn't contain a valid partition table