coolsnowwolf/lede #3281: K3编译失败

taoqixfz, 2020-02-23:
sed -i "s/Installed-Time: .*/Installed-Time: 1582380527/" /home/taoqixfz/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/usr/lib/opkg/status
rm -rf /home/taoqixfz/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/tmp/*
rm -f /home/taoqixfz/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/usr/lib/opkg/lists/*
rm -f /home/taoqixfz/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/usr/lib/opkg/info/*.postinst*
rm -f /home/taoqixfz/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/var/lock/*.lock
rm -rf /home/taoqixfz/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/boot
find /home/taoqixfz/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/ -mindepth 1 -execdir touch -hcd "@1582380527" "{}" +
find: The relative path 'Files/WindowsApps/CanonicalGroupLimited.Ubuntu16.04onWindows_1604.2019.523.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/taoqixfz/lede'
package/Makefile:108: recipe for target '/home/taoqixfz/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_install' failed
make[1]: *** [/home/taoqixfz/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/taoqixfz/lede'
/home/taoqixfz/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

EnnawYang, 2020-02-23:
似乎是windows的 $PATH问题 之前的issue好像有相关解决办法，可以自己搜下

taoqixfz, 2020-02-23:
wsl不知怎么多次编译失败，这次的还是第一次遇到，不知道wsl应该装Microsoft Store里面的Ubuntu还是Ubuntu16.04还是Ubuntu18.04

发自大磕碜的iPhone

Aes64X, 2020-02-23:
> wsl不知怎么多次编译失败，这次的还是第一次遇到，不知道wsl应该装Microsoft Store里面的Ubuntu还是Ubuntu16.04还是Ubuntu18.04 发自大磕碜的iPhone
> […](#)

VMware Workstation Pro + Ubuntu 18.04 它不香吗。。。。。。win自带的那个坑不少

yiguihai, 2020-02-23:
我也是迫不得已用wsl编译的，然后也是出现你这种错误。不过我修改环境变量后解决了

taoqixfz, 2020-02-23:
我也知道香，问题是我这破平板电脑跑不动

Aes64X, 2020-02-24:
> 我也知道香，问题是我这破平板电脑跑不动
> […](#)

平板。。。。上次回复当我没说。。。。。（666666666）

taoqixfz, 2020-02-24:
编译前执行：export PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games"后再次运行了make V=s后编译成功
lf@teclast:~/lede/bin/targets/bcm53xx/generic$ ls
config.seed                                                      openwrt-bcm53xx-dlink-dir-885l-squashfs.bin
openwrt-bcm53xx-arris-sbr-ac1900p-squashfs.trx                   openwrt-bcm53xx-linksys-ea6500-v2-squashfs.trx
openwrt-bcm53xx-arris-sbr-ac3200p-squashfs.trx                   openwrt-bcm53xx-netgear-r6250-squashfs.chk
openwrt-bcm53xx-asus-rt-ac56u-squashfs.trx                       openwrt-bcm53xx-netgear-r6300-v2-squashfs.chk
openwrt-bcm53xx-asus-rt-ac68u-squashfs.trx                       openwrt-bcm53xx-netgear-r7000-squashfs.chk
openwrt-bcm53xx-asus-rt-ac87u-squashfs.trx                       openwrt-bcm53xx-netgear-r7900-squashfs.chk
openwrt-bcm53xx-asus-rt-n18u-squashfs.trx                        openwrt-bcm53xx-netgear-r8000-squashfs.chk
openwrt-bcm53xx-buffalo-wxr-1900dhp-squashfs.trx                 openwrt-bcm53xx-phicomm-k3.manifest
openwrt-bcm53xx-buffalo-wzr-1750dhp-squashfs.trx                 openwrt-bcm53xx-phicomm-k3-squashfs.trx
openwrt-bcm53xx-buffalo-wzr-600dhp2-squashfs.trx                 openwrt-bcm53xx-smartrg-sr400ac-squashfs.trx
openwrt-bcm53xx-buffalo-wzr-900dhp-squashfs.factory-DHP2-JP.bin  packages
openwrt-bcm53xx-buffalo-wzr-900dhp-squashfs.factory-DHP-EU.bin   sha256sums
openwrt-bcm53xx-buffalo-wzr-900dhp-squashfs.trx
但是我还有个问题，我当时选择的是K3，但是结果编译出来的是所有博通53XX的CPU的固件，请教一下大神们，怎么才能实现仅编译指定型号，这样的话我觉得会节省很多时间

closed 2020-03-03 by coolsnowwolf

Jinnrry, 2020-04-07:
使用wsl ubuntu18.04遇到同样问题，修改windows注册表解决了

https://forum.openwrt.org/t/bug-today-the-compilation-does-not-come-to-the-end/45199

PR #3496 (2020-03-03): CI: add g++-multilib to dependence
cpufreq这个感谢指正，但是我不会写2333
就不重新开一个pr了，close #3510 
顺便 close #3270 , close #3276 , close #3281 , close #3291 , close #3336 , close #3414 .