coolsnowwolf/lede #3282: 最新的c9f2c9c编译不过

Jennal, 2020-02-23:
```
install -m0755 /root/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc/runc /root/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/ipkg-x86_64/runc/usr/sbin/
install: cannot stat '/root/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc/runc': No such file or directory
Makefile:91: recipe for target '/root/lede/bin/packages/x86_64/packages/runc_1.0.0-rc8+91-3e425f80-1_x86_64.ipk' failed
make[3]: *** [/root/lede/bin/packages/x86_64/packages/runc_1.0.0-rc8+91-3e425f80-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/root/lede/feeds/packages/utils/runc'
time: package/feeds/packages/runc/compile#0.77#0.66#1.93
package/Makefile:111: recipe for target 'package/feeds/packages/runc/compile' failed
make[2]: *** [package/feeds/packages/runc/compile] Error 2
make[2]: Leaving directory '/root/lede'
package/Makefile:107: recipe for target '/root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'
/root/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```

Jennal, 2020-02-23:
没有 `/root/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc/runc` 这个目录。
`/root/lede/build_dir/target-x86_64_musl/runc-1.0.0-rc8+91-3e425f80/.go_work/build/src/github.com/opencontainers/runc/runc` 这个目录是有的，是不是多了一个 `runc` ？

feizaibufei, 2020-02-23:
删除runc-1.0.0-rc8+91-3e425f80

Jennal, 2020-02-23:
> 删除runc-1.0.0-rc8+91-3e425f80

怎么删？

Jennal, 2020-02-23:
不知道被什么依赖了，在menuconfig里面没法改，是built-in的 -*-

hyird, 2020-02-23:
> > 删除runc-1.0.0-rc8+91-3e425f80
> 
> 怎么删？

build_dir里面的

closed 2020-02-27 by coolsnowwolf

Jennal, 2020-02-27:
> > > 删除runc-1.0.0-rc8+91-3e425f80
> > 
> > 
> > 怎么删？
> 
> build_dir里面的

删除了以后，runc通过了，但是到wget的时候，又过不去了

```
make[3]: Entering directory '/root/lede/feeds/packages/net/wget'
mkdir -p /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /root/lede/tmp/.root-copy.flock -c 'cp -fpR /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/.pkgdir/wget/. /root/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /root/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.wget_installed
mkdir -p /root/lede/bin/targets/x86/64/packages /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/ipkg-x86_64/wget/CONTROL /root/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/ipkg-x86_64/wget/usr/bin
install -m0755 /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/src/wget /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/ipkg-x86_64/wget/usr/bin/wget-ssl
find /root/lede/build_dir/target-x86_64_musl/wget-ssl/wget-1.20.3/ipkg-x86_64/wget -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package wget is missing dependencies for the following libraries:
libcrypto.so.1.0.0
libssl.so.1.0.0
Makefile:109: recipe for target '/root/lede/bin/packages/x86_64/packages/wget_1.20.3-2_x86_64.ipk' failed
make[3]: *** [/root/lede/bin/packages/x86_64/packages/wget_1.20.3-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/root/lede/feeds/packages/net/wget'
time: package/feeds/packages/wget/ssl/compile#0.78#0.65#1.49
package/Makefile:111: recipe for target 'package/feeds/packages/wget/compile' failed
make[2]: *** [package/feeds/packages/wget/compile] Error 2
make[2]: Leaving directory '/root/lede'
package/Makefile:107: recipe for target '/root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'
/root/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```