coolsnowwolf/lede #3283: 如何禁用dockerman安装的qBittorrent下载BT时触发机场的审计规则？

jy3344511, 2020-02-23:
之前大多用的SSR的，没被封过号，最近使用的两家v2ray，全部触发BT而被封号。

以下是v2ray的审计规则：
禁用 BT | 防止版权争议 | BitTorrent protocol
禁用 BT 2 | 防止版权争议 | (torrent\|\.torrent\|peer_id=\|info_hash\|get_peers\|find_node\|BitTorrent\|announce_peer\|announce\.php\?passkey=)

开启SSR Plus+时的使用环境：
1.软路由作为旁路由，IP为192.168.0.180，容器qbittorrent网络为bridge: 172.17.0.2。

2.在SSR Plus+的访问控制中，禁止IP地址:192.168.0.180、172.17.0.2走代理。其实这一步，不确定是否有用，但也设置了一下。

3.需要代理的端口：仅常用端口（不走P2P流量到代理），只要是开启SSR Plus+，一直都用的这个。

4.运行模式：GFW列表模式、绕过中国大陆IP模式，均测试过，都会触发机场的审计规则。备注：从未在开启全局代理的情况下，使用qBittorrent进行BT下载。

测试过，只要选择--仅常用端口（不走P2P流量到代理），无论是GFW列表模式还是绕过中国大陆IP模式，使用qbittorrent下载BT时，比如下载了10个G的流量，去机场网站刷新后发现，已使用的流量数值并未出现变化。然后关闭qbittorrent，打开油管看一个8K的视频，在此期间不停的刷新机场的网站，会发现已使用的流量数值，明显出现变化。可以发现下载BT的流量其实并没有走代理，但就是会触发机场设置的审计规则，而被封号，怎么解决？？？？

qbittorrent默认是使用6881的监听端口，SSR Plus+能否增加--禁止端口--走代理的功能呢？

怎么禁止qBittorrent触发BT的审计规则，有知道的说下，谢谢。

lloyd18, 2020-02-23:
BT未走代理还封号这是机场的问题，找机场去啊

jy3344511, 2020-02-23:
> BT未走代理还封号这是机场的问题，找机场去啊

机场主说触发了BT审计，不管，再者，即便解封，难道每次使用qBittorrent，都要把SSR Plus+关掉？其实很多BT种子，下载的速度是达不到满速的，也就是在这个时候，宽带没有使用的部分也是很多的，一边下载，一边看油管、上网的话，是很正常的事情，这个怎么解决呢？

其中的一家机场，在审计的规则里，也只写了个禁止BT，具体的规则也没写，即便写了，也不知道是什么，然后是禁止测速的也有，问过一些人，说是机场审计的太变态了，但有中转、价格合适的，不好找啊。现在又换回原来的SSR用了，墙高的时候大部分都用不了。但原来的SSR，不管是docker上跑qBittorrent，还是电脑主机开BitComet、迅雷下载，只要软路由上设置了--仅常用端口（不走P2P流量到代理），无论是GFW列表模式还是绕过中国大陆IP模式，下载1T、10T的东西，它就是不消耗机场的流量，第2步，禁止IP地址:192.168.0.180、172.17.0.2走代理这个，都不用设置，随便下载啥事没有，我也无语啊。

9l, 2020-02-23:
机场的问题

R3pl4c3r, 2020-02-23:
你可以专门整一个下载机，然后在访问控制里面输入不走代理的LAN IP，这样就不走代理了

bcd61, 2020-02-25:
说了这么多，你下BT难道不禁用Tracker？不禁用Tracker，有的URL还在GFWlist里，不是一样碰撞规则？

jy3344511, 2020-02-26:
> Tracker

没开自动更新Tracker服务器列表，手动添加了一些Tracker。

closed 2020-02-28 by jy3344511

fhbyljj, 2021-06-01:
我情况跟你一样，甚至本机IP不代理也没辙，我怀疑是Tracker问题

JJFJJML, 2022-03-04:
macvlan给docker独立IP，然后SrcIPCIDR direct

laaArmy, 2022-03-16:
之前我也是这样，薯条
明明群晖不走代理了，但还是触发了BT审查
后来我直接换了个机场
听说打开SFE会导致这个不走代理失效

CodeDreamfy, 2022-05-16:
> macvlan给docker独立IP，然后SrcIPCIDR direct

对 docker安装openwrt等旁路由单独一个macvlan，你再装qb应该就不影响了