coolsnowwolf/lede #3284: 强烈建议内置dnscrypt-proxy2，(地址: https://github.com/DNSCrypt/dnscrypt-proxy )作为ssrp自带的dns方案取代默认的pdnsd。固件自带的luci-app-dnscrypt-proxy是一个非常老的版本，几乎废无法使用。

Jasonsun77, 2020-02-23:
强烈建议内置dnscrypt-proxy2，(地址: https://github.com/DNSCrypt/dnscrypt-proxy )作为ssrp自带的dns方案取代默认的pdnsd。固件自带的luci-app-dnscrypt-proxy是一个非常老的版本，几乎废无法使用。

每次新升级以后都要先开ssrp内置pdnsd模式（因为不开代理根本没法下）然后用opkg安装dnscrypt-proxy2的ipk包（某些架构下还会出错，需要手动下载binary替换），然后配置好改ssrp为5335模式再重启。

只要做好分流，个人觉得smartDNS这种恨不得全世界都知道你在访问什么网站的多线查询是没有必要的，而加密解析则是至关重要，并且也是未来ESNI安全性的基石。且不说隐私问题，明文dns查询极易被代理提供商的一些莫名其妙的过滤规则误杀（如查询含有torrent关键字的域名）

个人采用的分流方案：
1. ssrp使用大陆IP白名单+5335作为dns解析的方式
2. dnscyrpt-proxy2配置监听在5335，仅使用DoH协议向cloudflare/adguard服务器查询（这些https的查询请求全部会被ssrp代理转发），并屏蔽ipv6的结果（ssrp能否支持同时连接v4 和v6两台代理服务器并分别处理不同流量呢）
3. iptables劫持所有目标为53端口的流量到路由器的53端口（这个默认就有了）
4. 53端口使用默认的dnsmasq作为dns服务，上游设置为127.0.0.1#5335，配置中添加`conf-dir=/etc/dnsmasq.d` 并在/etc/dnsmasq.d/中放入这个 https://github.com/felixonmars/dnsmasq-china-list/blob/master/accelerated-domains.china.conf 文件（并通过脚本自动更新）

这样实现了凡是大陆的和在大陆有cdn的以及返回AAAA记录的域名全部直连查询并直连访问，其余的全部走代理查询，且结果在大陆以外的走代理访问（IP+域名的双白名单机制）

_Originally posted by @hineH in https://github.com/coolsnowwolf/lede/issues/2551#issuecomment-568382313_

要是能出一个视频讲解就好了

0x5826, 2020-02-23:
其实就是把目前lean包里的那个替换掉新的这个咯

hcym, 2020-02-23:
用官方编译过，不太会用，不过passwall的dns转发有点意思，就是其它的地方不如大雕的好使

SharerMax, 2020-02-23:
欢迎PR

hyird, 2020-02-23:
大陆域名匹配走114 dns   其余走 dnscyrpt-proxy2?

wgjwillis, 2020-02-23:
有个luci-app ：[https://github.com/peter-tank/luci-app-dnscrypt-proxy2]，但关键问题：不会设置。

wgjwillis, 2020-02-23:
有个luci-app : [(https://github.com/peter-tank/luci-app-dnscrypt-proxy2) ]关键问题：不会设置。

hcym, 2020-02-24:
这个东西没有armv8，怪事

SharerMax, 2020-02-24:
#2604

FoxDring, 2020-02-24:
请教如何自己编译dnscrypt-proxy2.ipk

bcd61, 2020-02-25:
SmartDNS目前是开源状态，而且同样具备 DoT DoH 查询功能

closed 2020-03-06 by coolsnowwolf