coolsnowwolf/lede #3289: 访问jd.com 卡在解析misc.360buyimg.com好几秒

mjyhj, 2020-02-23:
关了SSR和adbyby还是一样。
PC开SSR全局代理 访问则秒开

hxlls, 2020-02-23:
我测试没问题，秒开，最近DNS 好像污染有点严重，前几天还发生过两广地区狗东解析到北京服务器去了，你换几个国内的DNS看看，然后打倒冷场王。。。。。。。。

mjyhj, 2020-02-23:
我在DHCP/DNS的LUCI里 ，直接点一下保存。马上就秒开了。
然后重启路由，又卡住，手动重启dnsmasq没用，又去LUCI里点一下保存 又可以秒开了

mjyhj, 2020-02-23:
> 我测试没问题，秒开，最近DNS 好像污染有点严重，前几天还发生过两广地区狗东解析到北京服务器去了，你换几个国内的DNS看看，然后打倒冷场王。。。。。。。。

不止我一个是这样，广东的好几个都一样

hxlls, 2020-02-23:
这次修改了不少规则，很多不兼容的，最好不保存设置升级，我遇到过保存设置升级就出问题，还有那个smartdns 和现在55R 好像有冲突，同时开启基本炸，测试了几次都如此，CPU直接爆炸，可能是ttl 延时等设置造成的，你自己测试下咯

hxlls, 2020-02-23:
如果是保存设置就秒开，那可能就是DNS 或者去广告规则上有问题，因为luci 可能保存时候清除了DNS缓存，那时候就回秒开，但是DNS 后面进行缓存时候，就不能秒开，规则加上去了

mjyhj, 2020-02-23:
> 如果是保存设置就秒开，那可能就是DNS 或者去广告规则上有问题，因为luci 可能保存时候清除了DNS缓存，那时候就回秒开，但是DNS 后面进行缓存时候，就不能秒开，规则加上去了

不是的，只要保存了一次，就可以一直秒开。但是重启了就会复现，adbyby和SSR里的去广告我都有关掉，还是一样的。
这个问题大概是半个月到一个月前开始这样

hxlls, 2020-02-23:
额，这个现象我倒是没有出现过 ，因为不懂具体使用的影响DNS 的软件情况，是不是做了2级的DNS 服务缓冲，所以还真不好猜测是哪里问题，我只能说，最新的源码，全新编译出来的，目前没碰到这种情况，我刚才测试了10多次，包括重启电脑和路由后再打开JD.COM 没发现卡住的现象，无法重现，有点难判断

mjyhj, 2020-02-23:
> 额，这个现象我倒是没有出现过 ，因为不懂具体使用的影响DNS 的软件情况，是不是做了2级的DNS 服务缓冲，所以还真不好猜测是哪里问题，我只能说，最新的源码，全新编译出来的，目前没碰到这种情况，我刚才测试了10多次，包括重启电脑和路由后再打开JD.COM 没发现卡住的现象，无法重现，有点难判断

原来确实过了一段时间后，又开始这样卡了

yangsongli, 2020-02-25:
我最近也 原因未知 我的百度直接解析到香港 360.com 直接解析到了美国 今天不保存配置重新刷openwrt解决

mjyhj, 2020-03-05:
> 我最近也 原因未知 我的百度直接解析到香港 360.com 直接解析到了美国 今天不保存配置重新刷openwrt解决

看来是东莞电信的锅
https://www.v2ex.com/t/645634

closed 2020-03-06 by coolsnowwolf