coolsnowwolf/lede #3291: 上个版本编译正常，今天更新代码出现新的问题。

feizaibufei, 2020-02-23:
Collected errors:
 * check_data_file_clashes: Package ntfs-3g wants to install file /home/openwrt/lede/build_dir/target-aarch64_generic_musl/root-armvirt/sbin/mount.ntfs
	But that file is already provided by package  * antfs-mount
 * opkg_install_cmd: Cannot install package automount.
 * check_data_file_clashes: Package ntfs-3g wants to install file /home/openwrt/lede/build_dir/target-aarch64_generic_musl/root-armvirt/sbin/mount.ntfs
	But that file is already provided by package  * antfs-mount
 * opkg_install_cmd: Cannot install package ntfs-3g.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/home/openwrt/lede'
make[1]: *** [package/Makefile:108: /home/openwrt/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
make: *** [/home/openwrt/lede/include/toplevel.mk:227：world] 错误 2

AmadeusGhost, 2020-02-23:
不选上`antfs-mount`，或者`rm .config`重新选

d975xbx, 2020-02-23:
[image]
找不到的话，打开.config
is not set然后注释掉

feizaibufei, 2020-02-23:
多谢

you-mu, 2020-02-24:
个人经验
编译前 rm -rf ./tmp
如果报错不明朗,我就make clean , 然后再开始.

FoxDring, 2020-02-26:
这是antfs-mount和ntfs-3g冲突，选前者。但如果ntfs-3g被锁定，需要取消Extra packages ---> automount，再回到Utilities ---> Filesystem 就可以取消ntfs-3g了。

feizaibufei, 2020-02-27:
多谢@FoxDring

txdywy, 2020-02-27:
同样问题，赞

PR #3496 (2020-03-03): CI: add g++-multilib to dependence
cpufreq这个感谢指正，但是我不会写2333
就不重新开一个pr了，close #3510 
顺便 close #3270 , close #3276 , close #3281 , close #3291 , close #3336 , close #3414 .
closed 2020-03-03 by coolsnowwolf