coolsnowwolf/lede #3295: Win10/Ubuntu编译报错不安全

lymatthew, 2020-02-23:
find /home/xxx/lean-lede/build_dir/target-x86_64_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1582293501" "{}" +
find: The relative path 'Files/WindowsApps/CanonicalGroupLimited.Ubuntu18.04onWindows_1804.2019.522.0_x64__79rhkp1fndgsc' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/xxx/lean-lede'
package/Makefile:108: recipe for target '/home/xxx/lean-lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/xxx/lean-lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/xxx/lean-lede'
/home/ly/lean-lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

大神好！最近两天编译报错，显示是环境变量不安全。用的是Win10下的Ubuntu，2月20日之前都没有问题，怀疑跟adjust build script (#3235)这条对编译脚本的修改有关，麻烦大神看一下，谢谢！

AmadeusGhost, 2020-02-23:
不关我事啊，same as https://github.com/coolsnowwolf/lede/issues/3251#issuecomment-589911815

Aes64X, 2020-02-23:
> find /home/xxx/lean-lede/build_dir/target-x86_64_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1582293501" "{}" +
> find: The relative path [...]

VMware Workstation Pro 它不香吗。。。。。干嘛都用Win10自带的。。。。

dolphinpaopao, 2020-02-23:
个人使用经验，Windows10 自带的Ubuntu编译出问题的几率高于虚拟机安装Ubuntu原版，VB+18.04 server一直非常稳定编译固件。

ckxt, 2020-02-23:
不要用WSL了  VMWARE虚拟机里玩吧

lymatthew, 2020-02-23:
@AmadeusGhost 抱歉&谢谢！
找到原因了，是WSL与windows互交互，但CSDN的那个教程是错的。正确的办法是改 /etc/wsl.conf

[interop]
enabled=false # enable launch of Windows binaries; default is true
appendWindowsPath=false # append Windows path to $PATH variable; default is true

WSL的好处是直接在 Win里用笔记本编辑就可以了。。 @dolphinpaopao @Aes64X @ckxt

Aes64X, 2020-02-23:
> @AmadeusGhost 抱歉&谢谢！
> 找到原因了，是WSL与windows互交互，但CSDN的那个教程是错的。正确的办法是改 /etc/wsl.conf
> 
> [interop]
> enabled=false # enable launch of Windows binaries; default is true
> appendWindowsPath=false # append Windows path to $PATH variable; default is true
> 
> WSL的好处是直接在 Win里用笔记本编辑就可以了。。 @dolphinpaopao @Aes64X @ckxt

你说的是笔记本编辑？还是我理解错了？

[image]

Aes64X, 2020-02-23:
我懂了

AmadeusGhost, 2020-02-23:
查了下，这个问题是windows环境变量里面有空格导致的
```
PATH=$(echo "$PATH" | sed -e 's/:\/mnt.*//g') # strip out problematic Windows %PATH% imported var
```

helianling, 2020-02-23:
编译前在Bash里面执行：export PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games"

closed 2020-02-24 by lymatthew

mkcczh, 2020-10-02:
楼主你好，我也是有同样的编译出错，WSL2，但是我在\\wsl$\Ubuntu-18.04\etc下找不到wsl.conf，请问是这个路径吗？

lymatthew, 2020-10-03:
> 楼主你好，我也是有同样的编译出错，WSL2，但是我在\wsl$\Ubuntu-18.04\etc下找不到wsl.conf，请问是这个路径吗？

是的。你要新建一个。

mkcczh, 2020-10-03:
谢谢楼主，编译成功了！