coolsnowwolf/lede #3298: 概况-网络不能显示多WAN状态

dengchan0416, 2020-02-23:
我的宽带能四拨,在IPv4 WAN 状态里只能显示其中一个,怎样才能显示全部接口.
[image]

AmadeusGhost, 2020-02-24:
IPv4 WAN只会显示一个（物理上上的wan口），其它的是mwan

9l, 2020-02-24:
你这不就是显示的四个么，四个 pppoe

82kg, 2020-02-24:
> 你这不就是显示的四个么，四个 pppoe

他说是的在首页显示 所有WAN口信息

tty228, 2020-02-24:
首页显示的话，替换掉 `/usr/lib/lua/luci/view/admin_status` 目录文件
[index.zip](https://github.com/coolsnowwolf/lede/files/4243363/index.zip)

dengchan0416, 2020-02-24:
@AmadeusGhost 四个中其中还有一个是物理的WAN口联通的,一条电信3多拨,另一条联通1拨

AmadeusGhost, 2020-02-24:
> @AmadeusGhost 四个中其中还有一个是物理的WAN口联通的,一条电信3多拨,另一条联通1拨

我知道你意思，但是IPv4 WAN这个地方是显示物理wan口的，其他是显示到mwan那，此项目默认屏蔽了mwan3在首页上的显示，楼上的dalao已经给你替换文件了

dengchan0416, 2020-02-25:
@tty228 请问是这样吗?
[image]

wafm0708, 2020-02-29:
> 我的宽带能四拨,在IPv4 WAN 状态里只能显示其中一个,怎样才能显示全部接口.
> [image]
> 
> [image]

请问多线多拨的插件是什么

closed 2020-03-06 by coolsnowwolf

elfive, 2021-03-25:
> /usr/lib/lua/luci/view/admin_status

谢谢，不过你那个在最新的openwrt和luci下面似乎不会在首页->概况->IPv4 WAN 状态那里显示多WAN的信息。
按照这个[提交](https://github.com/openwrt/luci/commit/a13748d4145edd3edc70e462714a115f7c99470e)修改可以试试。