coolsnowwolf/lede #3299: RTL8169驱动无法加载

Kotori-0912, 2020-02-23:
今天git pull后编译出来的版本无法加载RTL8169驱动，编译时已经确定kmod-r8169是勾选上的，编译过程无任何错误，不保留配置更新后没有看到加载8169驱动信息，使用ifconfig查看网络信息也只有lo接口无eth接口

hxlls, 2020-02-23:
[image]
默认选择就已经有驱动

Kotori-0912, 2020-02-23:
> [image]
> 默认选择就已经有驱动

我用KVM虚拟机直通RTL8169网卡但是无法加载，上一个版本是能正常加载
[image]
[image]

hyird, 2020-02-24:
r8169-firmware | 20191215-1 | 22827 | RealTek RTL8169 firmware
你这个装了么

Kotori-0912, 2020-02-24:
> r8169-firmware | 20191215-1 | 22827 | RealTek RTL8169 firmware
> 你这个装了么

r8169-firmware这个有安装到

Kotori-0912, 2020-02-24:
删除全部源代码重新拉去后再次编译，情况解决，可能与上次强行覆盖导致源代码有问题

closed 2020-02-24 by Kotori-0912