coolsnowwolf/lede #3300: 發現SSR+一個小bug，V2RAY設定裏的Transport=TCP, Camouflage Type的設定是無效的

sammimk830, 2020-02-23:
我今天剛試了一下用V2RAY的TLS+HTTP 伪装，但是發現在設定裏面是沒有了HTTP這個的設定的。
/var/etc/v2-ssr-retcp.json
/var/etc/v2-ssr-reudp.json
找到問題在下面這個設定里少了一段tcpSettings，加上後就沒有問題了，不知道是否真的漏了。
/usr/share/shadowsocksr/genv2config.lua
以下是我自己加的，成功可以用，但不知道為什麼，headers裏只能加一個host，再加多一個參數就會產生不了json了。
tcpSettings = (server.transport == "tcp") and (server.tcp_guise == "http") and (server.http_path ~= nil or server.http_host ~= nil) and {
  header = (server.http_host ~= nil) and {
    type = server.tcp_guise,
    request = {
      version = "1.1",
                method = "GET",
      path = server.http_path,
      headers = {
        host = server.http_host
      }
    }
  }  or nil,
} or nil,

[image]

SharerMax, 2020-02-24:
path 和 host 对应的数据是个数组结构，多个host你得拆成数组结构填入

另外 TLS + http -> https ? 
为什么不直接http2

sammimk830, 2020-02-24:
1. 現在我輸入的不是多個host，只有一個，只是想加多一些參數就不行，可看下圖
2. 我現在的V2RAY走的是TCP+TLS，是真實的TLS來的，我只是需要PATH來幫我去分走的背後走去哪，意思和WSS是差不多的，但是我現在是可以WSS或者TCP+TLS，而所謂的HTTP，只是用來告訴PATCH去分配去哪個服務器。
3. http2還是在HTTP層面上走，我現在是走得比較快的TCP+TLS呢。不知道能不能明白。
[image]

closed 2020-03-06 by coolsnowwolf