coolsnowwolf/lede #3317: 编译选中最新的解锁网易云音乐，就会报错，

silime, 2020-02-25:
[image]
编译到node-v12.15.0就报错了

mjyhj, 2020-02-25:
更新依赖
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib linux-libc-dev:i386

不行再make clean

waterun, 2020-02-26:
> 更新依赖
> sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs [...]

Mj牛，我也折腾了半天，还以为是bug

closed 2020-02-26 by coolsnowwolf