coolsnowwolf/lede #3324: 无法更新订阅节点，请大神指点迷津

abclevel, 2020-02-25:
2020-02-25 13:54:03 Shadowsocks/ShadowsocksR 2 Threads Started!
2020-02-25 13:50:13 ShadowsocksR No Problem.
2020-02-25 13:39:04 Shadowsocks/ShadowsocksR 2 Threads Started!
2020-02-25 13:39:03 重启服务成功
2020-02-25 13:39:03 发生错误, 正在恢复服务
	[C]: ?
	/usr/share/shadowsocksr/subscribe.lua:383: in main chunk
	[C]: in function 'xpcall'
	/usr/share/shadowsocksr/subscribe.lua:295: in function </usr/share/shadowsocksr/subscribe.lua:246>
	/usr/share/shadowsocksr/subscribe.lua:133: in function 'processData'
	/usr/share/shadowsocksr/subscribe.lua:385: in function </usr/share/shadowsocksr/subscribe.lua:383>
2020-02-25 13:39:03 stack traceback:
2020-02-25 13:39:03 /usr/share/shadowsocksr/subscribe.lua:133: attempt to index local 'info' (a nil value)

2020-02-25 13:39:03 (76.8 MB/s) - written to stdout [1789/1789]

abclevel, 2020-02-25:
下午用wget -O /usr/share/shadowsocksr/subscribe.lua https://github.com/coolsnowwolf/lede/tree/master/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/subscribe.lua，更新节点，日志有以下错误：
020-02-25 16:37:21 Shadowsocks/ShadowsocksR 2 Threads Started! 2020-02-25 16:37:16 重启服务成功 2020-02-25 16:37:16 发生错误, 正在恢复服务  [C]: ?  /usr/share/shadowsocksr/subscribe.lua:383: in main chunk  [C]: in function 'xpcall'  /usr/share/shadowsocksr/subscribe.lua:295: in function </usr/share/shadowsocksr/subscribe.lua:246>  /usr/share/shadowsocksr/subscribe.lua:228: in function 'processData'  /usr/share/shadowsocksr/subscribe.lua:385: in function </usr/share/shadowsocksr/subscribe.lua:383> 2020-02-25 16:37:16 stack traceback: 2020-02-25 16:37:16 /usr/share/shadowsocksr/subscribe.lua:228: attempt to concatenate field 'server_port' (a nil value)  2020-02-25 16:37:16 (248 KB/s) - written to stdout [440915]

abclevel, 2020-02-25:
> 2020-02-25 13:54:03 Shadowsocks/ShadowsocksR 2 Threads Started!
> 2020-02-25 13:50:13 ShadowsocksR No Problem.
> 2020-02-25 13:39:04 Shadowsocks/ShadowsocksR 2 Threads Started!
> 2020-02-25 13:39:03 [...]

closed 2020-02-25 by abclevel

lichunjiang32, 2020-10-24:
2020-10-24 22:45:54 /usr/share/shadowsocksr/subscribe.lua:33: attempt to index local 'full' (a nil value)
2020-10-24 22:45:54 stack traceback:
        /usr/share/shadowsocksr/subscribe.lua:436: in function </usr/share/shadowsocksr/subscribe.lua:434>
        /usr/share/shadowsocksr/subscribe.lua:33: in function 'split'
        /usr/share/shadowsocksr/subscribe.lua:182: in function 'processData'
        /usr/share/shadowsocksr/subscribe.lua:341: in function </usr/share/shadowsocksr/subscribe.lua:294>
        [C]: in function 'xpcall'
        /usr/share/shadowsocksr/subscribe.lua:434: in main chunk
        [C]: ?

lichunjiang32, 2020-10-24:
2020-10-24 22:45:54 /usr/share/shadowsocksr/subscribe.lua:33: attempt to index local 'full' (a nil value)
2020-10-24 22:45:54 stack traceback:
        /usr/share/shadowsocksr/subscribe.lua:436: in function </usr/share/shadowsocksr/subscribe.lua:434>
        /usr/share/shadowsocksr/subscribe.lua:33: in function 'split'
        /usr/share/shadowsocksr/subscribe.lua:182: in function 'processData'
        /usr/share/shadowsocksr/subscribe.lua:341: in function </usr/share/shadowsocksr/subscribe.lua:294>
        [C]: in function 'xpcall'
        /usr/share/shadowsocksr/subscribe.lua:434: in main chunk
        [C]: ?
请问怎么解决的？

iceshu, 2021-02-03:
我也是 如何解决的啊

Soupure, 2024-01-06:
怎么解决的哇