coolsnowwolf/lede #3327: Trojan客户端时断时续

WellsWang, 2020-02-25:
利用VPS建立了一个Trojan服务器，使用ShadowSocksR Plus+插件中的trojan支持连接服务器，发先服务器状态侦测时断时续，导致无法连接，日志如下：
2020-02-25 15:45:12 Welcome to trojan 1.14.0 Started!
2020-02-25 15:44:13 Try restart current server.
2020-02-25 15:44:10 Welcome to trojan 1.14.0 Started!
2020-02-25 15:43:47 Current server error, try to switch another server.
2020-02-25 15:42:37 Try restart current server.
2020-02-25 15:42:34 Welcome to trojan 1.14.0 Started!
2020-02-25 15:42:10 Current server error, try to switch another server.
2020-02-25 15:41:00 ShadowsocksR No Problem.
2020-02-25 15:39:57 Try restart current server.
2020-02-25 15:39:54 Welcome to trojan 1.14.0 Started!
2020-02-25 15:39:31 Current server error, try to switch another server.
2020-02-25 15:38:21 Try restart current server.
2020-02-25 15:38:18 Welcome to trojan 1.14.0 Started!
2020-02-25 15:37:55 Current server error, try to switch another server.
2020-02-25 15:36:45 Try restart current server.
2020-02-25 15:36:42 Welcome to trojan 1.14.0 Started!
2020-02-25 15:36:18 Current server error, try to switch another server.
2020-02-25 15:35:08 Try restart current server.
2020-02-25 15:35:05 Welcome to trojan 1.14.0 Started!

事实上，日志报告服务器有问题的时候，通过Windows Trojan客户端连接都是正常，服务器的HTTP/HTTPS端口也能正常连接。
尝试过Azure/AWS，在世界不同地区设置过服务器。服务器是否打开BBR都如此。
是否有方法调试或查找到原因？

SharerMax, 2020-02-25:
你可能需要更新

WellsWang, 2020-02-25:
@SharerMax 您是指服务器端还是lede的客户端插件？

SharerMax, 2020-02-25:
> @SharerMax 您是指服务器端还是lede的客户端插件？

Lede

WellsWang, 2020-02-25:
@SharerMax 我github上应该是上周clone的最新的……，是否插件要更新？
插件在哪有更新呢？
谢谢！

SharerMax, 2020-02-26:
自己编译，如果没记错的话，1.14.1已经有一段时间了

txdywy, 2020-02-26:
感觉有可能是dns的问题

SharerMax, 2020-02-26:
@txdywy 恩，dns的可能性有点高，毕竟现在各种乱用三方DNS的教程有点多

WellsWang, 2020-02-26:
@txdywy 您好，我试过服务器名写IP地址，也是一样，另外，如果是DNS问题，我认为使用shadowsocks也会有同样问题，但是shadowsocks使用没有这个问题。或者两者服务器探测机制不同？

另外，DNS是否有设置的建议？

txdywy, 2020-02-26:
我的版本目前是感觉pdns偶尔有些不正常，还在研究。感觉trojan自己还是比较稳

WellsWang, 2020-02-29:
昨晚重新pull了最新的代码重新编译后正常了……
可能是1.14.0的bug?现在是1.14.1

WellsWang, 2020-03-02:
发现并未解决问题，当我重新启动 Shadowsocks Plus+服务，又出现了同样的问题，而且无论如何改配置，都回不去了，即使出厂化设置也不行……

只有编译好后第一次运行配置后那一天是好的，我重新启动了下Shadowsocks Plus+服务就坏事了。

我尝试了：
关闭端口映射
关闭IPV6
服务器地址写IP地址
DNS服务器更换为海外服务器
关闭BBR加速和DNS解析加速

都没有用……

真的不知道怎么做，是否有调试的方法，让我知道是什么原因也好。

服务器测试fail的时候我用windows客户端直连服务器代理都是好的。

或者是否有另外Workaround的方法，比如我在路由器上安装trojan-cli的linux版本，他会启动一个本地的Socks5代理，在LEDE上是否有类似浏览器的SwitchyOmega插件来绕过中国IP或这GFWList来使用本地Socks代理的插件?

lipwah, 2020-03-02:
> 发现并未解决问题，当我重新启动 Shadowsocks Plus+服务，又出现了同样的问题，而且无论如何改配置，都回不去了，即使出厂化设置也不行……
>
> 只有编译好后第一次运行配置后那一天是好的，我重新启动了下Shadowsocks Plus+服务就坏事了。
>
> 我尝试了：
> 关闭端口映射
> 关闭IPV6
> 服务器地址写IP地址
> DNS服务器更换为海外服务器
> 关闭BBR加速和DNS解析加速 [...]

我也遇到这样问题，用着Shadowsocks Plus+服务一阵子，突然又连接不上，刚开始以为节点问题，但是我换回win 上的客户端可以正常连接，感觉Shadowsocks Plus+很不稳定

henrayluo, 2020-03-03:
也遇到同样的问题了：
2020-03-03 11:18:35 Try restart current server.
2020-03-03 11:18:30 Welcome to trojan 1.14.1 Started!
2020-03-03 11:18:04 Current server error, try to switch another server.
2020-03-03 11:06:38 ShadowsocksR No Problem.
2020-03-03 10:55:31 ShadowsocksR No Problem.
2020-03-03 10:44:24 Try restart current server.
2020-03-03 10:44:19 Welcome to trojan 1.14.1 Started!
2020-03-03 10:43:53 Current server error, try to switch another server.
2020-03-03 10:32:27 Try restart current server.
2020-03-03 10:32:23 Welcome to trojan 1.14.1 Started!
2020-03-03 10:31:55 Current server error, try to switch another server.
2020-03-03 10:20:29 ShadowsocksR No Problem.
2020-03-03 10:09:18 ShadowsocksR No Problem.
2020-03-03 09:58:11 ShadowsocksR No Problem.
2020-03-03 09:47:04 ShadowsocksR No Problem.
2020-03-03 09:35:56 ShadowsocksR No Problem.
2020-03-03 09:24:49 ShadowsocksR No Problem.
2020-03-03 09:13:41 ShadowsocksR No Problem.
2020-03-03 09:02:34 ShadowsocksR No Problem.
2020-03-03 08:51:27 ShadowsocksR No Problem.
2020-03-03 08:40:19 ShadowsocksR No Problem.
2020-03-03 08:29:12 ShadowsocksR No Problem.
2020-03-03 08:18:05 ShadowsocksR No Problem.
2020-03-03 08:06:57 ShadowsocksR No Problem.
2020-03-03 07:55:50 ShadowsocksR No Problem.
2020-03-03 07:44:42 ShadowsocksR No Problem.
2020-03-03 07:33:35 ShadowsocksR No Problem.
2020-03-03 07:22:28 ShadowsocksR No Problem.
2020-03-03 07:11:21 ShadowsocksR No Problem.
2020-03-03 07:00:13 ShadowsocksR No Problem.
2020-03-03 06:49:05 ShadowsocksR No Problem.
2020-03-03 06:37:58 ShadowsocksR No Problem.
2020-03-03 06:26:51 ShadowsocksR No Problem.
2020-03-03 06:15:44 ShadowsocksR No Problem.
2020-03-03 06:04:36 ShadowsocksR No Problem.
2020-03-03 05:53:28 ShadowsocksR No Problem.
2020-03-03 05:42:21 ShadowsocksR No Problem.
2020-03-03 05:31:14 ShadowsocksR No Problem.
2020-03-03 05:20:07 ShadowsocksR No Problem.
2020-03-03 05:08:59 ShadowsocksR No Problem.
2020-03-03 04:57:52 ShadowsocksR No Problem.
2020-03-03 04:46:44 ShadowsocksR No Problem.
2020-03-03 04:35:37 ShadowsocksR No Problem.
2020-03-03 04:24:30 ShadowsocksR No Problem.
2020-03-03 04:13:22 ShadowsocksR No Problem.
2020-03-03 04:02:15 ShadowsocksR No Problem.
2020-03-03 03:51:08 ShadowsocksR No Problem.
2020-03-03 03:40:00 ShadowsocksR No Problem.
2020-03-03 03:28:53 ShadowsocksR No Problem.
2020-03-03 03:17:45 ShadowsocksR No Problem.
2020-03-03 03:06:38 ShadowsocksR No Problem.
2020-03-03 02:55:31 ShadowsocksR No Problem.
2020-03-03 02:44:24 ShadowsocksR No Problem.
2020-03-03 02:33:16 ShadowsocksR No Problem.
2020-03-03 02:22:09 ShadowsocksR No Problem.
2020-03-03 02:11:01 Try restart current server.
2020-03-03 02:10:56 Welcome to trojan 1.14.1 Started!
2020-03-03 02:10:28 Current server error, try to switch another server.
2020-03-03 01:59:02 ShadowsocksR No Problem.
2020-03-03 01:47:55 ShadowsocksR No Problem.
2020-03-03 01:36:48 ShadowsocksR No Problem.
2020-03-03 01:25:40 ShadowsocksR No Problem.
2020-03-03 01:14:33 ShadowsocksR No Problem.
2020-03-03 01:03:25 ShadowsocksR No Problem.

closed 2020-03-06 by coolsnowwolf