coolsnowwolf/lede #3328: [feature request] luci-ssr-plus: forward dns to ssr server instead of using pdnsd or localhost:5335

kmahyyg, 2020-02-25:
We use SNI Proxy to unlock the oversea multimedia streaming service, which need the dns resolve process to be done on the remote server instead of any other third-party DNS or client local DNS.

Could you please help me add an option to just forward the dns query to server instead of using any other dns service software? Thanks.

coolsnowwolf, 2020-02-26:
it already worked like this way

kmahyyg, 2020-02-26:
then how should I set the `work_mode` and `pdnsd_enable` ?

kmahyyg, 2020-02-26:
I have checked about the script... it seems like in `mangle` chain, the 53 port is `return` instead of `tproxy`.

`https://github.com/kmahyyg/lede/blob/master/package/lean/luci-app-ssr-plus/root/usr/bin/ssr-rules#L188`

z826540272, 2020-02-26:
it already worked like this way

kmahyyg, 2020-02-26:
> it already worked like this way

I mean that the domain name which user wanna access to should be sent to remote **Shadowsocks** server and let the server resolve it. The current behaviour is just resolve using 8.8.8.8 if pdnsd enabled.

closed 2020-02-26 by kmahyyg

Limezy, 2020-03-05:
@kmahyyg would you mind explaining here how you did ? I have exactly the same question. Thanks !

kmahyyg, 2020-03-05:
I just don't want to argue, since I have no choice but this software are proved and tested will not send the DNS request to remote shadowsocks server based on the streaming server is failed to detect correct location when I use this software. And also, the shell script in this software also proved this. So I changed to another router firmware.

Another third party Ref here: https://blog.skk.moe/post/how-proxy-client-cause-netflix-unlock-to-fail/

I do urge the developer fix this problem.

istanleyyaw, 2020-11-26:
I want to have the same request.
I agree with kmahyyg that the DNS query is at local openwrt when I use trojan as server. Here is how I identify it:
1. I set as GLobal Mode in SSR
2. on the trojan server, I didn't see any request coming to DNS IP
3. when I browse some webiste having CDN service, checking the server log, it access CDN node with near to my original location instead of near to the trojan server location, make it very slow.

kmahyyg, 2020-11-26:
This issue is closed, please do not reply here, since I won't offer any helpful information here. Currently, I just grab a `clash` binary from padavan and write init&iptables script myself. All things works fine now.