coolsnowwolf/lede #3330: 内置screen的可信性？

DongfeiSay, 2020-02-25:
一些情况下在screen下比较安全，opkg install screen却找不到软件包，或者有其他安装方法吗？

lloyd18, 2020-02-25:
root@OpenWrt:~# opkg install screen
Installing screen (4.7.0-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/screen_4.7.0-1_x86_64.ipk
Configuring screen.

DongfeiSay, 2020-02-25:
> root @ OpenWrt：〜＃opkg安装屏幕正在将
> 屏幕（4.7.0-1）安装到root ...正在
> 下载[https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/screen_4.7.0-1_x86_64。 ipk](https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/screen_4.7.0-1_x86_64.ipk)
> 配置屏幕。

我这里MIPS架构的是提示
root@Unknown:~# Collected errors:
-ash: Collected: not found
root@Unknown:~# opkg install screen
Unknown package 'screen'.
Collected errors:
 * opkg_install_cmd: Cannot install package screen.

DongfeiSay, 2020-02-25:
> root@OpenWrt:~# opkg install screen
> Installing screen (4.7.0-1) to root...
> Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/screen_4.7.0-1_x86_64.ipk
> Configuring screen.

我这边mips构架的是提示
Unknown package 'screen'.
Collected errors:
 * opkg_install_cmd: Cannot install package screen.

DongfeiSay, 2020-02-25:
> > root@OpenWrt:~# opkg install screen
> > Installing screen (4.7.0-1) to root...
> > Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/screen_4.7.0-1_x86_64.ipk
> > Configuring screen.
> 
> 我这边mips构架的是提示
Collected errors:
> Unknown package 'screen'.
> Collected errors:
> 
> * opkg_install_cmd: Cannot install package screen.

DongfeiSay, 2020-02-25:
> root@OpenWrt:~# opkg install screen
> Installing screen (4.7.0-1) to root...
> Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/screen_4.7.0-1_x86_64.ipk
> Configuring screen.

我这边misp构架的是提示
Unknown package 'screen'.
Collected errors:
 * opkg_install_cmd: Cannot install package screen.
root@Unknown:~# opkg install screen
Unknown package 'screen'.
Collected errors:
 * opkg_install_cmd: Cannot install package screen.

DongfeiSay, 2020-02-25:
> root@OpenWrt:~# opkg install screen
> Installing screen (4.7.0-1) to root...
> Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/packages/screen_4.7.0-1_x86_64.ipk
> Configuring screen.

我这边misp构架的是提示
Unknown package 'screen'.
Collected errors:
 * opkg_install_cmd: Cannot install package screen.

lphgor, 2020-02-25:
snapshots里面好像确实没有，从19.07里面找吧
https://openwrt.proxy.ustclug.org/releases/packages-19.07/

xinxijishuwyq, 2020-02-25:
编译进去啊。。

closed 2020-02-25 by DongfeiSay