coolsnowwolf/lede #3331: DDNS更新不成功

sunday229, 2020-02-25:
用了https://www.pubyun.com/的f3322域名。日志都提示更新成功了，可是ping域名始终更新不了。
 175847  info : PID '23765' received 'SIGHUP' at 2020-02-25 17:58
 175847  note : PID '23765' exit normal at 2020-02-25 17:58
 175847       : ************ ************** ************** **************
 175847  note : PID '29387' started at 2020-02-25 17:58
 175847       : ddns version  : 2.7.8-1
 175847       : uci configuration:
ddns.myddns_ipv4.domain='ABC.f3322.net'
ddns.myddns_ipv4.enabled='1'
ddns.myddns_ipv4.force_interval='8'
ddns.myddns_ipv4.force_ipversion='1'
ddns.myddns_ipv4.interface='wan'
ddns.myddns_ipv4.ip_network='wan'
ddns.myddns_ipv4.ip_source='network'
ddns.myddns_ipv4.lookup_host='ABC.f3322.net'
ddns.myddns_ipv4.password='***PW***'
ddns.myddns_ipv4.service_name='3322.org'
ddns.myddns_ipv4.username='root'
ddns.myddns_ipv4=service
 175848       : verbose mode  : 0 - run normal, NO console output
 175848       : check interval: 600 seconds
 175848       : force interval: 28800 seconds
 175848       : retry interval: 60 seconds
 175848       : retry counter : 0 times
 175848       : No old process
 175848       : last update: 2020-02-25 17:51
 175848       : Detect registered/public IP
 175848  note : Busybox nslookup - no support to 'force IP Version' (ignored)
 175848       : #> /usr/bin/nslookup ABC.f3322.net  >/var/run/ddns/myddns_ipv4.dat 2>/var/run/ddns/myddns_ipv4.err
 175848       : Registered IP 'X.X.X.131' detected
 175848  info : Starting main loop at 2020-02-25 17:58
 175848       : Detect local IP on 'network'
 175848       : Local IP 'X.X.X.240' detected on network 'wan'
 175848       : Update needed - L: 'X.X.X.240' <> R: 'X.X.X.131'
 175849       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err -4 --no-proxy 'http://root:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=ABC.f3322.net&myip=X.X.X.240'
 175849       : DDNS Provider answered:
nochg X.X.X.240
 175849  info : Update successful - IP 'X.X.X.240' send
 175849  info : Forced update successful - IP: 'X.X.X.240' send
 175849       : Waiting 600 seconds (Check Interval)
 175909  info : PID '29387' received 'SIGHUP' at 2020-02-25 17:59
 175909  note : PID '29387' exit normal at 2020-02-25 17:59
 175909       : ************ ************** ************** **************
 175909  note : PID '2692' started at 2020-02-25 17:59
 175909       : ddns version  : 2.7.8-1
 175909       : uci configuration:
ddns.myddns_ipv4.domain='ABC.f3322.net'
ddns.myddns_ipv4.enabled='1'
ddns.myddns_ipv4.force_interval='8'
ddns.myddns_ipv4.force_ipversion='1'
ddns.myddns_ipv4.interface='wan'
ddns.myddns_ipv4.ip_network='wan'
ddns.myddns_ipv4.ip_source='network'
ddns.myddns_ipv4.lookup_host='ABC.f3322.net'
ddns.myddns_ipv4.password='***PW***'
ddns.myddns_ipv4.service_name='3322.org'
ddns.myddns_ipv4.username='root'
ddns.myddns_ipv4=service
 175910       : verbose mode  : 0 - run normal, NO console output
 175910       : check interval: 600 seconds
 175910       : force interval: 28800 seconds
 175910       : retry interval: 60 seconds
 175910       : retry counter : 0 times
 175910       : No old process
 175910       : last update: 2020-02-25 17:58
 175910       : Detect registered/public IP
 175910  note : Busybox nslookup - no support to 'force IP Version' (ignored)
 175910       : #> /usr/bin/nslookup ABC.f3322.net  >/var/run/ddns/myddns_ipv4.dat 2>/var/run/ddns/myddns_ipv4.err
 175910       : Registered IP 'X.X.X.131' detected
 175910  info : Starting main loop at 2020-02-25 17:59
 175910       : Detect local IP on 'network'
 175910       : Local IP 'X.X.X.240' detected on network 'wan'
 175910       : Update needed - L: 'X.X.X.240' <> R: 'X.X.X.131'
 175910       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err -4 --no-proxy 'http://root:***PW***@members.3322.org/dyndns/update?system=dyndns&hostname=ABC.f3322.net&myip=X.X.X.240'
 175911       : DDNS Provider answered:
nochg X.X.X.240
 175911  info : Update successful - IP 'X.X.X.240' send
 175911  info : Forced update successful - IP: 'X.X.X.240' send
 175911       : Waiting 600 seconds (Check Interval)

lphgor, 2020-02-25:
先确认一下，直接ping IP能通吗？

sunday229, 2020-02-25:
ping IP没问题的 用IP测试端口映射都通的
 刚测试了下，又可以了 更新太慢了。日志提示都成功的，web页面一直反应不过来，刷了很多个版本都是这个问题，麻烦大佬测试一下。

sunday229, 2020-02-25:
175911  info : Update successful - IP 'X.X.X.240' send
175911  info : Forced update successful - IP: 'X.X.X.240' send

9l, 2020-02-25:
域名解析本来就是有延迟的

closed 2020-02-26 by coolsnowwolf