coolsnowwolf/lede #3333: nodejs 小米r3p 编译出错

mingxiaoyu, 2020-02-25:
错误信息
make[5]: *** [/home/runner/work/actionbuild/actionbuild/openwrt/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o] Error 1
tools/v8_gypfiles/v8_libbase.host.mk:148: recipe for target '/home/runner/work/actionbuild/actionbuild/openwrt/build_dir/target-mipsel_24kc_musl/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o' failed
make[4]: *** [node] Error 2
make[3]: ***

waterun, 2020-02-25:
acrh17 same issue

smous, 2020-02-25:
筋斗云编译一样的报错..

d975xbx, 2020-02-25:
newifi3 d2 云编译和本地编译一样报错

mingxiaoyu, 2020-02-25:
/usr/include/c++/7/type_traits:38:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>

wgedu, 2020-02-25:
node-v12.15.0的锅，挂全局，make download V=s 之后在编译

mingxiaoyu, 2020-02-25:
> node-v12.15.0的锅，挂全局，make download V=s 之后在编译

我都是github action的。。。应该不会有这个问题吧

hubbylei, 2020-02-25:
In file included from ../deps/v8/src/base/bits.h:9:0,
                 from ../deps/v8/src/base/bits.cc:5:
/usr/include/c++/7/type_traits:38:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
同样的错误:(

closed 2020-02-26 by coolsnowwolf