coolsnowwolf/lede #3334: 编译了个FriendlyElec NanoPi R2S固件用官方仓库加上lean的luci app仓库有人要吗？

programmejava, 2020-02-25:
[image]
折腾了两天，用官方代码加lean的代码

lloyd18, 2020-02-25:
有没有相关教程 我也想搞个杂交

kl9800, 2020-02-25:
求教程

programmejava, 2020-02-25:
http://wiki.friendlyarm.com/wiki/index.php/How_to_Build_FriendlyWrt/zh
官方编译wiki 拉好代码把lean的仓库拉好update -a install -a之后复制package/lean包去官方目录下编译，仔细阅读官方的教程
主是要这个依赖不好搞，各种奇怪错误一会儿有，一会儿没有

kl9800, 2020-02-25:
> http://wiki.friendlyarm.com/wiki/index.php/How_to_Build_FriendlyWrt/zh
> 官方编译wiki 拉好代码把lean的仓库拉好update -a install -a之后复制package/lean包去官方目录下编译，仔细阅读官方的教程
> 主是要这个依赖不好搞，各种奇怪错误一会儿有，一会儿没有
> 不能选argon和netgear rosy主题 不然一直卡luci界面 不能编译luci-app-guestwifi 否则进luci界面出错

谢谢大佬！！！

lloyd18, 2020-02-25:
> http://wiki.friendlyarm.com/wiki/index.php/How_to_Build_FriendlyWrt/zh
> 官方编译wiki 拉好代码把lean的仓库拉好update -a install -a之后复制package/lean包去官方目录下编译，仔细阅读官方的教程
> 主是要这个依赖不好搞，各种奇怪错误一会儿有，一会儿没有
> 不能选argon和netgear rosy主题 不然一直卡luci界面 不能编译luci-app-guestwifi 否则进luci界面出错

搞过几次 解决完依赖还是编译出错 一直没成功

Aes64X, 2020-02-25:
这插件。。。看着眼晕 - -#

programmejava, 2020-02-25:
> > http://wiki.friendlyarm.com/wiki/index.php/How_to_Build_FriendlyWrt/zh
> > 官方编译wiki 拉好代码把lean的仓库拉好update -a install -a之后复制package/lean包去官方目录下编译，仔细阅读官方的教程
> > 主是要这个依赖不好搞，各种奇怪错误一会儿有，一会儿没有
> > 不能选argon和netgear rosy主题 不然一直卡luci界面 不能编译luci-app-guestwifi 否则进luci界面出错
> 
> 搞过几次 解决完依赖还是编译出错 一直没成功

它的目录结构这样的friendlywrt-rk3328/friendlywrt 你源码配置好先在friendlywrt这个目录里make -j8 V=s （8是线程数量，出错后就改为1看看出错在哪里）先编译如果编译通过再到上层文件夹friendlywrt-rk3328里执行官方的../build.sh nanopi_r2s.mk编译生成SD卡固件  在内层编译主要是排错。

lloyd18, 2020-02-25:
> 它的目录结构这样的friendlywrt-rk3328/friendlywrt 你源码配置好先在friendlywrt这个目录里make -j8 V=s （8是线程数量，出错后就改为1看看出错在哪里）先编译如果编译通过再到上层文件夹friendlywrt-rk3328里执行官方的../build.sh nanopi_r2s.mk编译生成SD卡固件 在内层编译主要是排错。

我是要编译x86的

mayh520, 2020-02-26:
求教程

closed 2020-03-06 by coolsnowwolf

adamgonglj, 2020-04-08:
> [image]
> 折腾了两天，用官方代码加lean的代码

有教程吗？我弄了几次，编译通不过。