coolsnowwolf/lede #3336: psg1208编译错误

untitledtxt, 2020-02-25:
configure: error: Oops, mp_limb_t doesn't seem to work
Makefile:35: recipe for target '/home/antony/lede/build_dir/host/gmp-6.1.2/.configured' failed
make[3]: *** [/home/antony/lede/build_dir/host/gmp-6.1.2/.configured] Error 1
make[3]: Leaving directory '/home/antony/lede/tools/gmp'
time: tools/gmp/compile#6.29#3.33#9.75
tools/Makefile:161: recipe for target 'tools/gmp/compile' failed
make[2]: *** [tools/gmp/compile] Error 2
make[2]: Leaving directory '/home/antony/lede'
tools/Makefile:159: recipe for target '/home/antony/lede/staging_dir/target-mipsel_24kc_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynnynynnyyyynynyyy' failed
make[1]: *** [/home/antony/lede/staging_dir/target-mipsel_24kc_musl/stamp/.tools_compile_yynyynnyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynnynynnyyyynynyyy] Error 2
make[1]: Leaving directory '/home/antony/lede'
/home/antony/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

修改过luci列表，make clean 之后重新执行也是同样报错

lizhenglin2002, 2020-02-25:
我这X86的经常编译了第一次，第二次就不行了，显示ERROR2

royalwei, 2020-02-26:
export FORCE_UNSAFE_CONFIGURE=1

royalwei, 2020-02-26:
出错输入这个再继续编译就行

lizhenglin2002, 2020-02-26:
> 错误输入这个再继续编译就行

麻烦帮我看看这是什么原因的错误
[image]

royalwei, 2020-02-26:
> > 错误输入这个再继续编译就行
> 
> 麻烦帮我看看这是什么原因的错误
> [image]
删掉.config重新编译，因为大雕某次更新中返回了ntfs-3g，不再用antfs-mount了

lizhenglin2002, 2020-02-26:
> > > 错误输入这个再继续编译就行
> > 
> > 
> > 麻烦帮我看看这是什么原因的错误
> > [image]
> > 删掉.config重新编译，因为大雕某次更新中返回了ntfs-3g，不再用antfs-mount了

请问是在哪里删除

royalwei, 2020-02-26:
> > > > 错误输入这个再继续编译就行
> > > 
> > > 
> > > 麻烦帮我看看这是什么原因的错误
> > > [image]
> > > 删掉.config重新编译，因为大雕某次更新中返回了ntfs-3g，不再用antfs-mount了
> 
> 请问是在哪里删除

rm .config

lizhenglin2002, 2020-02-26:
> > > > > 错误输入这个再继续编译就行
> > > > 
> > > > 
> > > > 麻烦帮我看看这是什么原因的错误
> > > > [image]
> > > > 删掉.config重新编译，因为大雕某次更新中返回了ntfs-3g，不再用antfs-mount了
> > 
> > 
> > 请问是在哪里删除
> 
> rm .config

好的谢谢

PR #3496 (2020-03-03): CI: add g++-multilib to dependence
cpufreq这个感谢指正，但是我不会写2333
就不重新开一个pr了，close #3510 
顺便 close #3270 , close #3276 , close #3281 , close #3291 , close #3336 , close #3414 .
closed 2020-03-03 by coolsnowwolf