coolsnowwolf/lede #3337: 编译的n1固件V2RAY设置不能上网

feizaibufei, 2020-02-25:
[image]
[image]
[image]
[image]

feizaibufei, 2020-02-25:
Tue Feb 25 20:36:21 2020 daemon.notice netifd: Interface 'lan' is now down
Tue Feb 25 20:36:21 2020 daemon.info avahi-daemon[3991]: Registering new address record for fe80::fe7c:2ff:fe89:13d5 on br-lan.*.
Tue Feb 25 20:36:21 2020 daemon.info avahi-daemon[3991]: Withdrawing address record for 192.168.32.137 on br-lan.
Tue Feb 25 20:36:21 2020 daemon.info avahi-daemon[3991]: Leaving mDNS multicast group on interface br-lan.IPv4 with address 192.168.32.137.
Tue Feb 25 20:36:21 2020 daemon.info avahi-daemon[3991]: Interface br-lan.IPv4 no longer relevant for mDNS.
Tue Feb 25 20:36:21 2020 kern.info kernel: [  156.489796] br-lan: port 2(wlan0) entered disabled state
Tue Feb 25 20:36:21 2020 kern.info kernel: [  156.494556] br-lan: port 1(eth0) entered disabled state
Tue Feb 25 20:36:21 2020 daemon.info avahi-daemon[3991]: Interface br-lan.IPv6 no longer relevant for mDNS.
Tue Feb 25 20:36:21 2020 daemon.info avahi-daemon[3991]: Leaving mDNS multicast group on interface br-lan.IPv6 with address fd76:e2bb:728::1.
Tue Feb 25 20:36:21 2020 kern.info kernel: [  156.506179] device eth0 left promiscuous mode
Tue Feb 25 20:36:21 2020 kern.info kernel: [  156.511275] br-lan: port 1(eth0) entered disabled state
Tue Feb 25 20:36:21 2020 kern.debug kernel: [  156.547491] meson8b-dwmac c9410000.ethernet eth0: phy link down rgmii/1Gbps/Full
Tue Feb 25 20:36:21 2020 kern.info kernel: [  156.547494] meson8b-dwmac c9410000.ethernet eth0: Link is Down
Tue Feb 25 20:36:21 2020 daemon.info avahi-daemon[3991]: Withdrawing address record for fe80::fe7c:2ff:fe89:13d5 on br-lan.
Tue Feb 25 20:36:21 2020 daemon.info avahi-daemon[3991]: Withdrawing address record for fd76:e2bb:728::1 on br-lan.
Tue Feb 25 20:36:21 2020 kern.info kernel: [  156.569886] device wlan0 left promiscuous mode
Tue Feb 25 20:36:21 2020 kern.info kernel: [  156.576807] br-lan: port 2(wlan0) entered disabled state
Tue Feb 25 20:36:21 2020 daemon.notice netifd: Interface 'lan' is disabled
Tue Feb 25 20:36:21 2020 daemon.notice netifd: Interface 'lan' has link connectivity loss
Tue Feb 25 20:36:21 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.707471] meson8b-dwmac c9410000.ethernet eth0: PHY [0.2009087f:00] driver [RTL8211F Gigabit Ethernet]
Tue Feb 25 20:36:22 2020 kern.debug kernel: [  156.717320] meson8b-dwmac c9410000.ethernet eth0: phy: setting supported 00,00000000,000062ff advertising 00,00000000,000062ff
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.822831] meson8b-dwmac c9410000.ethernet eth0: No Safety Features support found
Tue Feb 25 20:36:22 2020 kern.warn kernel: [  156.837616] meson8b-dwmac c9410000.ethernet eth0: PTP not supported by HW
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.844494] meson8b-dwmac c9410000.ethernet eth0: configuring for phy/rgmii link mode
Tue Feb 25 20:36:22 2020 kern.debug kernel: [  156.852246] meson8b-dwmac c9410000.ethernet eth0: phylink_mac_config: mode=phy/rgmii/Unknown/Unknown adv=00,00000000,000062ff pause=10 link=0 an=1
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.852929] br-lan: port 1(eth0) entered blocking state
Tue Feb 25 20:36:22 2020 kern.debug kernel: [  156.854331] meson8b-dwmac c9410000.ethernet eth0: phy link down rgmii/Unknown/Unknown
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.857722] br-lan: port 1(eth0) entered disabled state
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.862927] device eth0 entered promiscuous mode
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.868755] br-lan: port 2(wlan0) entered blocking state
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.873515] br-lan: port 2(wlan0) entered disabled state
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.878671] device wlan0 entered promiscuous mode
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.885124] br-lan: port 2(wlan0) entered blocking state
Tue Feb 25 20:36:22 2020 kern.info kernel: [  156.889847] br-lan: port 2(wlan0) entered forwarding state
Tue Feb 25 20:36:22 2020 daemon.notice netifd: Interface 'lan' is enabled
Tue Feb 25 20:36:22 2020 daemon.notice netifd: Interface 'lan' is setting up now
Tue Feb 25 20:36:22 2020 daemon.info avahi-daemon[3991]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.32.137.
Tue Feb 25 20:36:22 2020 daemon.info avahi-daemon[3991]: New relevant interface br-lan.IPv4 for mDNS.
Tue Feb 25 20:36:22 2020 daemon.info avahi-daemon[3991]: Registering new address record for 192.168.32.137 on br-lan.IPv4.
Tue Feb 25 20:36:22 2020 daemon.info avahi-daemon[3991]: Joining mDNS multicast group on interface br-lan.IPv6 with address fd76:e2bb:728::1.
Tue Feb 25 20:36:22 2020 daemon.info avahi-daemon[3991]: New relevant interface br-lan.IPv6 for mDNS.
Tue Feb 25 20:36:22 2020 daemon.info avahi-daemon[3991]: Registering new address record for fd76:e2bb:728::1 on br-lan.*.
Tue Feb 25 20:36:22 2020 daemon.notice netifd: Interface 'lan' is now up
Tue Feb 25 20:36:22 2020 daemon.notice netifd: Network device 'eth0' link is down
Tue Feb 25 20:36:23 2020 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Tue Feb 25 20:36:23 2020 daemon.err odhcpd[1594]: Failed to send to ff02::1%lan@br-lan (Bad file descriptor)
Tue Feb 25 20:36:23 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Feb 25 20:36:23 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
[... cut ...]
Tue Feb 25 20:49:44 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Feb 25 20:49:44 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Feb 25 20:49:46 2020 daemon.err uhttpd[3502]: grep: write error: Broken pipe
Tue Feb 25 20:49:50 2020 daemon.err uhttpd[3502]: grep: write error: Broken pipe
Tue Feb 25 20:49:52 2020 daemon.err uhttpd[3502]: grep: write error: Broken pipe
Tue Feb 25 20:49:55 2020 daemon.err uhttpd[3502]: grep: write error: Broken pipe

coolsnowwolf, 2020-02-26:
源代码不支持 N1

closed 2020-02-26 by coolsnowwolf

hcym, 2020-02-26:
完全没有任何问题啊😯😯

feizaibufei, 2020-02-26:
搞不懂，关闭N1固件V2RAY 换回电脑版V2RAY又正常。