coolsnowwolf/lede #3340: 更新一下mt76的驱动，解决5g wifi 的mcu timeout问题

j3l11234, 2020-02-25:
最近总是出现5g wifi不稳定的问题，严重时需要重启才能恢复。日志如下
```
[ 3464.467785] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 3464.473266] mt76x2e 0000:02:00.0: Build: 1
[ 3464.477408] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 3464.506137] mt76x2e 0000:02:00.0: Firmware running!
[ 3464.516288] ieee80211 phy1: Hardware restart was requested
[ 3465.681540] pppoe-wan: renamed from ppp0
[ 5195.065102] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[ 5254.336677] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 5254.342567] mt76x2e 0000:02:00.0: Build: 1
[ 5254.346945] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 5254.374953] mt76x2e 0000:02:00.0: Firmware running!
[ 5254.384994] ieee80211 phy1: Hardware restart was requested
[ 5257.836586] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 5257.842067] mt76x2e 0000:02:00.0: Build: 1
[ 5257.846233] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 5257.874780] mt76x2e 0000:02:00.0: Firmware running!
[ 5257.885019] ieee80211 phy1: Hardware restart was requested
[ 8801.616826] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[ 8842.988925] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 8842.994732] mt76x2e 0000:02:00.0: Build: 1
[ 8842.999694] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[ 8843.027341] mt76x2e 0000:02:00.0: Firmware running!
[ 8843.037840] ieee80211 phy1: Hardware restart was requested
[10614.733895] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[10669.735793] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[10669.743209] mt76x2e 0000:02:00.0: Build: 1
[10669.747319] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[10669.773286] mt76x2e 0000:02:00.0: Firmware running!
[10669.783317] ieee80211 phy1: Hardware restart was requested
```

看了openwrt那边的驱动，他们已经解决了这个问题
https://github.com/openwrt/mt76/commit/17ecf076254216a4011e93ab621fc651b047df9f

L大有空更新一下驱动呀。

YanminZhou, 2020-02-26:
对的 我也遇到了.....L大求修复呀

j3l11234, 2020-02-26:
> 对的 我也遇到了.....L大求修复呀

我昨天试着自己匹配到最新的mt76的驱动，发现有api不匹配，仔细一看发现是 mac80211模块不匹配，官方是5.2rc，L大之前不知道为何把它降级到了4.19，所以我也不知道怎么办了。

YanminZhou, 2020-02-26:
> > 对的 我也遇到了.....L大求修复呀
> 
> 我昨天试着自己匹配到最新的mt76的驱动，发现有api不匹配，仔细一看发现是 mac80211模块不匹配，官方是5.2rc，L大之前不知道为何把它降级到了4.19，所以我也不知道怎么办了。

我又发现有点不一样...我这边没有报错mcu timeout，但是同样是5g wifi很不稳定，不确定问题出在哪里
`[13410.055447] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[13410.061025] mt76x2e 0000:02:00.0: Build: 1
[13410.065279] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[13410.093571] mt76x2e 0000:02:00.0: Firmware running!
[13410.103741] ieee80211 phy1: Hardware restart was requested
[21512.441010] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[21512.446496] mt76x2e 0000:02:00.0: Build: 1
[21512.450660] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[21512.479265] mt76x2e 0000:02:00.0: Firmware running!
[21512.489518] ieee80211 phy1: Hardware restart was requested
`

coolsnowwolf, 2020-02-26:
无线驱动不用 rc 版本，等官方push

j3l11234, 2020-02-26:
> 无线驱动不用 rc 版本，等官方push
 
好的，L大辛苦。
我看官方现在也是这个版本，那就等官方也升级吧。
```
PKG_SOURCE_DATE:=2020-01-04
PKG_SOURCE_VERSION:=8a78567983a16869b77a0254b4917027df4a7ad9
PKG_MIRROR_HASH:=0cc48bc7093b99dc7f43a5a83a3a00d205709326f529b91b0b14fe0d89cb7
```
https://github.com/openwrt/openwrt/blob/v19.07.1/package/kernel/mt76/Makefile

j3l11234, 2020-02-26:
> 我又发现有点不一样...我这边没有报错mcu timeout，但是同样是5g wifi很不稳定，不确定问题出在哪里
> 
@YanminZhou 
5gwifi卡的直接问题出在`ieee80211 phy1: Hardware restart was requested`
我们两都被重启了，不同的是我重启的40s前有一个`mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out`而你没有。 我是昨天下午用L大的最新版本编译的。你呢？

YanminZhou, 2020-02-26:
> @YanminZhou
> 5gwifi卡的直接问题出在`ieee80211 phy1: Hardware restart was requested`
> 我们两都被重启了，不同的是我重启的40s前有一个`mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out`而你没有。 我是昨天下午用L大的最新版本编译的。你呢？

我目前在用2月4日下午编译的版本，正在尝试编译最新的版本。

knightridersimon, 2020-02-28:
我用mac80211 5.4 和最新MT76，在openwrt trunk和x-wrt上试过了，无线起不来，报告
ipv6 addrconf(netdev_up) wlan0 link is not ready ，懒得折腾了，等更新吧

closed 2020-03-06 by coolsnowwolf