coolsnowwolf/lede #3346: 单网口虚拟机LAN/WAN共用DHCP失效

rnm330, 2020-02-26:
我现在是使用单网口win10虚拟机桥接网络安装的lede，lan和wan共用eth0。

目前遇到的问题是各种使用都ok但是dhcp功能在第一次使用的时候正常，重启一次过后会开始失效，无法指派ip。

查询了很多地方和很多方法都无效，手动在客户机上指派是可以的，dnsmas的67端口也是正常的，非常费解。

目前的解决方法是安装adguardhome，其中自带了一个hdcp服务器，目前来指派ip，也算是一石二鸟吧，但是是什么导致dnsmasp无法指派ip还是想求教一下，很不理解。

makepoloSeton, 2020-02-27:
日志给一个，wan口是什么方式联网，pppoe，dhcp分配，静态ip？网络拓扑给一个，接的上级设备有无开启dhcp功能

rnm330, 2020-02-27:
拓扑如下
[image]

网络内没有其他设备开启dhcp
日志如下：
`Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.277530] pci 0000:00:15.7:   bridge window [mem 0xfb800000-0xfb8fffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.277666] pci 0000:00:15.7:   bridge window [mem 0xe5e00000-0xe5efffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.277855] pci 0000:00:16.0: PCI bridge to [bus 0b]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.277955] pci 0000:00:16.0:   bridge window [io  0x5000-0x5fff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.278091] pci 0000:00:16.0:   bridge window [mem 0xfd300000-0xfd3fffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.278265] pci 0000:00:16.0:   bridge window [mem 0xe7900000-0xe79fffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.278468] pci 0000:00:16.1: PCI bridge to [bus 0c]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.278569] pci 0000:00:16.1:   bridge window [io  0x9000-0x9fff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.278706] pci 0000:00:16.1:   bridge window [mem 0xfcf00000-0xfcffffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.278843] pci 0000:00:16.1:   bridge window [mem 0xe7500000-0xe75fffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.279032] pci 0000:00:16.2: PCI bridge to [bus 0d]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.279134] pci 0000:00:16.2:   bridge window [io  0xd000-0xdfff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.279270] pci 0000:00:16.2:   bridge window [mem 0xfcb00000-0xfcbfffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.279406] pci 0000:00:16.2:   bridge window [mem 0xe7100000-0xe71fffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.279596] pci 0000:00:16.3: PCI bridge to [bus 0e]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.279717] pci 0000:00:16.3:   bridge window [mem 0xfc700000-0xfc7fffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.279853] pci 0000:00:16.3:   bridge window [mem 0xe6d00000-0xe6dfffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.280043] pci 0000:00:16.4: PCI bridge to [bus 0f]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.280164] pci 0000:00:16.4:   bridge window [mem 0xfc300000-0xfc3fffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.280302] pci 0000:00:16.4:   bridge window [mem 0xe6900000-0xe69fffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.280505] pci 0000:00:16.5: PCI bridge to [bus 10]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.280633] pci 0000:00:16.5:   bridge window [mem 0xfbf00000-0xfbffffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.280776] pci 0000:00:16.5:   bridge window [mem 0xe6500000-0xe65fffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.280996] pci 0000:00:16.6: PCI bridge to [bus 11]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.281134] pci 0000:00:16.6:   bridge window [mem 0xfbb00000-0xfbbfffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.281283] pci 0000:00:16.6:   bridge window [mem 0xe6100000-0xe61fffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.281509] pci 0000:00:16.7: PCI bridge to [bus 12]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.281649] pci 0000:00:16.7:   bridge window [mem 0xfb700000-0xfb7fffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.281803] pci 0000:00:16.7:   bridge window [mem 0xe5d00000-0xe5dfffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.282035] pci 0000:00:17.0: PCI bridge to [bus 13]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.282233] pci 0000:00:17.0:   bridge window [io  0x6000-0x6fff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.282414] pci 0000:00:17.0:   bridge window [mem 0xfd200000-0xfd2fffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.282550] pci 0000:00:17.0:   bridge window [mem 0xe7800000-0xe78fffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.282755] pci 0000:00:17.1: PCI bridge to [bus 14]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.282855] pci 0000:00:17.1:   bridge window [io  0xa000-0xafff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.282992] pci 0000:00:17.1:   bridge window [mem 0xfce00000-0xfcefffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.283127] pci 0000:00:17.1:   bridge window [mem 0xe7400000-0xe74fffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.283315] pci 0000:00:17.2: PCI bridge to [bus 15]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.283414] pci 0000:00:17.2:   bridge window [io  0xe000-0xefff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.283550] pci 0000:00:17.2:   bridge window [mem 0xfca00000-0xfcafffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.283685] pci 0000:00:17.2:   bridge window [mem 0xe7000000-0xe70fffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.283871] pci 0000:00:17.3: PCI bridge to [bus 16]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.283991] pci 0000:00:17.3:   bridge window [mem 0xfc600000-0xfc6fffff]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.284126] pci 0000:00:17.3:   bridge window [mem 0xe6c00000-0xe6cfffff 64bit pref]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.284311] pci 0000:00:17.4: PCI bridge to [bus 17]
Thu Feb 27 13:44:34 2020 kern.info kernel: [    0.284436] pci 0000:00:17.4:   bridge window [mem 0xfc200000-0xfc2fffff]
[... cut ...]
Thu Feb 27 13:53:44 2020 daemon.err uhttpd[2594]: The called action terminated with an exception:
Thu Feb 27 13:53:44 2020 daemon.err uhttpd[2594]: Name does not resolve
Thu Feb 27 13:53:44 2020 daemon.err uhttpd[2594]: stack traceback:
Thu Feb 27 13:53:44 2020 daemon.err uhttpd[2594]: 	[C]: in function 'assert'
Thu Feb 27 13:53:44 2020 daemon.err uhttpd[2594]: 	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
Thu Feb 27 13:53:44 2020 daemon.err uhttpd[2594]: 	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>
`

closed 2020-03-06 by coolsnowwolf