coolsnowwolf/lede #3349: Flow Offloading 转发加速不要随便开

R3pl4c3r, 2020-02-26:
许多UP主/Youtuber在介绍Turbo ACC加速时都会让你把四个都开开。然而，我有公网IP，当我打开这个Flow Offloading 转发加速后，我发现我从外部能访问到我的软路由的其他服务。路由器全端口暴露在公网上，防火墙失效。关闭之后，外部访问被防火墙阻止。仅端口转发生效

R3pl4c3r, 2020-02-26:
这玩意开开太他妈吓人了

royalwei, 2020-02-26:
在搞笑吗？我公网ip开了咋没啥事，你贴出你被攻击的依据？

royalwei, 2020-02-26:
开了WAN口访问，设置强点密码，就算有尝试登陆都被拒绝掉，和Turbo ACC没关系，建议自己多用搜索引擎和学习英语

R3pl4c3r, 2020-02-26:
> 在搞笑吗？我公网ip开了咋没啥事，你贴出你被攻击的依据？

https://www.v2ex.com/t/608821?p=4
架设web服务停宽带签保证书了解一下

R3pl4c3r, 2020-02-26:
> > 在搞笑吗？我公网ip开了咋没啥事，你贴出你被攻击的依据？
> 
> https://www.v2ex.com/t/608821?p=4
> 架设web服务停宽带签保证书了解一下

根据宽带接入服务协议，第六条，第四款
六、协议的中止、解除与终止
（一）甲方有下列情形之一，乙方可以中止本协议（暂停提供服务）：
1、提供不真实的客户信息资料；
2、未经乙方同意，擅自在已装的通信线路上装、移（室内移机除外）各种终端设备及
复用设备或转让租用权的；或将宽带以任何方式提供给第三方使用或用作于其他运营商的
违规互联；
3、擅自改变客户类型及使用性质的；或擅自改变宽带安装地址的；
4、利用乙方提供的拨号宽带网络开设 WEB 服务的；

royalwei, 2020-02-26:
> > 在搞笑吗？我公网ip开了咋没啥事，你贴出你被攻击的依据？
> 
> https://www.v2ex.com/t/608821?p=4
> 架设web服务停宽带签保证书了解一下

害怕就不用

royalwei, 2020-02-26:
> 根据宽带接入服务协议，第六条，第四款
> 六、协议的中止、解除与终止
> （一）甲方有下列情形之一，乙方可以中止本协议（暂停提供服务）：
> 1、提供不真实的客户信息资料；
> 2、未经乙方同意，擅自在已装的通信线路上装、移（室内移机除外）各种终端设备及
> 复用设备或转让租用权的；或将宽带以任何方式提供给第三方使用或用作于其他运营商的
> 违规互联；
> 3、擅自改变客户类型及使用性质的；或擅自改变宽带安装地址的； [...]

你想表达什么？这和固件什么关系，和Turbo ACC什么关系，你自己开的WAN口访问和端口转发，
没人逼你开对吧？

R3pl4c3r, 2020-02-26:
> 你想表达什么？这和固件什么关系，和Turbo ACC什么关系，你自己开的WAN口访问和端口转发，
> 没人逼你开对吧？

我就是给提个醒，自己路由器上开了web服务最好把Turbo ACC的转发加速关了。否则一旦被ISP扫到会被封宽带，有些小白光按那些up主/youtuber的来，到时候宽带被封了人家可不负责。

royalwei, 2020-02-26:
Turbo ACC的原理只有dns加速需要把dns数据缓存在本地，其他不涉及外网，为什么会被isp扫到？

R3pl4c3r, 2020-02-26:
> Turbo ACC的原理只有dns加速需要把dns数据缓存在本地，其他不涉及外网，为什么会被isp扫到？

[image]
看图，Turbo ACC有三个加速，你说的那是DNS加速，我说的是转发加速

R3pl4c3r, 2020-02-26:
> > Turbo ACC的原理只有dns加速需要把dns数据缓存在本地，其他不涉及外网，为什么会被isp扫到？
> 
> [image]
> 看图，Turbo ACC有三个加速，你说的那是DNS加速，我说的是转发加速

转发加速会把路由器的所有端口转发到Wan口上去

royalwei, 2020-02-26:
不是这个原理，他是利用的算法，sfe是高通的开源引擎

R3pl4c3r, 2020-02-26:
> 不是这个原理，他是利用的算法，sfe是高通的开源引擎

但事实是我开了Flow Offloading，我并没有转发8096、8080端口，UPnP也没有看到相关端口，但是我用我的手机关闭WiFi，用4G就可以直接用8096和8080访问我路由器上的页面。关闭之后，就无法从外部访问了

royalwei, 2020-02-26:
我试试，是不是你们运营商开了这个端口

coolsnowwolf, 2020-02-26:
检查自己防火墙策略 如有这个问题请向openwrt 报告，因为测试过并不会

R3pl4c3r, 2020-02-26:
> 检查自己防火墙策略 如有这个问题请向openwrt 报告，因为测试过并不会

防火墙是默认设置，除了端口转发没有做任何改动
[image]

yangsongli, 2020-02-26:
开启Flow Offloading并不会把端口暴露在公网把 防火墙会拦截的 UPnP才会自动开端口

R3pl4c3r, 2020-02-26:
[image]
[image]
[image]
[image]
防火墙、Turbo ACC默认转发加速开启，UPnP未见8096、21、9999、264、8080、8388、8389端口，防火墙端口转发未开启
手机4G链接情况：
[image]
[image]
[image]
[image]
[image]
[image]
UPnP关闭后
[image]
手机依然可以连接
[image]
Flow Offloading关闭UPnP打开后
[image]
手机访问264/8096/9999失败
[image]
[image]
[image]

EnnawYang, 2020-02-26:
一句话带过：你是个例

R3pl4c3r, 2020-02-26:
更新一下，准确说是Flow Offloading转发加速会把docker的端口暴露在公网

Aes64X, 2020-02-26:
> 更新一下，准确说是Flow Offloading转发加速会把docker的端口暴露在公网，路由器固件内的app所使用的21，22，8080不受影响

老哥，我能理解你现在的心情，很兴奋，你认为你发现了一个惊天大秘密。但是。。。你这标题不合适在这里，用加速转发的人不在少数，几乎都用。你难道不该先搜索一下有没有以往的旧事吗？然后再来发问大家自己是什么问题。这里是issues，不是论坛。还有，加速转发是不碰端口的，你应该检查下你的防火墙FullNAT全转发设置和Docker我如果没记错他有个选项是是否将Docker暴露给公网。

z826540272, 2020-02-26:
设计缺陷,应该算是bug

R3pl4c3r, 2020-02-26:
> > 更新一下，准确说是Flow Offloading转发加速会把docker的端口暴露在公网，路由器固件内的app所使用的21，22，8080不受影响
> 
> 老哥，我能理解你现在的心情，很兴奋，你认为你发现了一个惊天大秘密。但是。。。你这标题不合适在这里，用加速转发的人不在少数，几乎都用。你难道不该先搜索一下有没有以往的旧事吗？然后再来发问大家自己是什么问题。这里是issues，不是论坛。还有，加速转发是不碰端口的，你应该检查下你的防火墙FullNAT全转发设置和Docker我如果没记错他有个选项是是否将Docker暴露给公网。

@Aes64X  最早是UPnP把8080弄出去了，从而造成变量过多，因此我并不清楚是什么造成的，现在弄清楚了，issue可以关了
解决方案：
#1756 以及[Docker与IPtables - 简书](https://www.jianshu.com/p/69d3ab177655)

closed 2020-02-26 by R3pl4c3r

R3pl4c3r, 2020-02-26:
> > 更新一下，准确说是Flow Offloading转发加速会把docker的端口暴露在公网，路由器固件内的app所使用的21，22，8080不受影响
> 
> 老哥，我能理解你现在的心情，很兴奋，你认为你发现了一个惊天大秘密。但是。。。你这标题不合适在这里，用加速转发的人不在少数，几乎都用。你难道不该先搜索一下有没有以往的旧事吗？然后再来发问大家自己是什么问题。这里是issues，不是论坛。还有，加速转发是不碰端口的，你应该检查下你的防火墙FullNAT全转发设置和Docker我如果没记错他有个选项是是否将Docker暴露给公网。

@Aes64X 现在所有网络上写docker image部署的还有docker官方都应该拖出去斩了，前者不看官方手册，后者的端口映射方式tm太反人类了。端口那块得弄成127.0.0.1:xxx:xxx，docker才不会在iptables上打洞。

whlsxl, 2020-03-28:
没记错的话，docker 会在iptables里加规则，把自己加到白名单里。
docker 给服务器用的，本来就是需要把服务暴露出来。
（记错的话，请告诉我