coolsnowwolf/lede #3353: acrh17编译失败

lqhorochi, 2020-02-26:
tools/v8_gypfiles/v8_base_without_compiler.host.mk:26: warning: ignoring old commands for target `/home/k/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Schema.h'
  touch /home/k/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/tools/v8_gypfiles/v8_version.stamp
  touch /home/k/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/tools/v8_gypfiles/v8_headers.stamp
  g++ -m32 -o /home/k/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o ../deps/v8/src/base/bits.cc '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-DV8_TARGET_ARCH_ARM' '-DCAN_USE_ARMV7_INSTRUCTIONS' '-DCAN_USE_VFP3_INSTRUCTIONS' '-DCAN_USE_VFP32DREGS' '-DCAN_USE_NEON' '-DV8_EMBEDDER_STRING="-node.16"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DV8_USE_SNAPSHOT' '-DV8_CONCURRENT_MARKING' '-DV8_EMBEDDED_BUILTINS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DBUILDING_V8_BASE_SHARED' '-DUSE_EABI_HARDFLOAT=1' -I/home/k/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I../deps/v8 -I../deps/v8/include  -pthread -Wno-unused-parameter -Wno-return-type -fno-strict-aliasing -m32 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/k/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/.deps//home/k/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o.d.raw   -c
In file included from ../deps/v8/src/base/bits.h:9:0,
                 from ../deps/v8/src/base/bits.cc:5:
/usr/include/c++/4.8/type_traits:38:28: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
                            ^
compilation terminated.
make[5]: *** [/home/k/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o] Error 1
make[5]: Leaving directory `/home/k/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out'
make[4]: *** [node] Error 2
make[4]: Leaving directory `/home/k/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0'
make[3]: *** [/home/k/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/.built] Error 2
make[3]: Leaving directory `/home/k/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#3.38#0.81#5.38
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory `/home/k/lede'
make[1]: *** [/home/k/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/k/lede'
make: *** [world] 错误 2

programmejava, 2020-02-26:
安装gcc-multilib g++-multilib  如果还不行就带上版本

lunatickochiya, 2020-02-26:
这是node的编译问题，错误日志很清楚，缺少某个文件

closed 2020-02-26 by coolsnowwolf