coolsnowwolf/lede #3355: 关于优化新版DNSMASQ IPV6过滤功能

wybb, 2020-02-26:
这个想法在去年年底我也考虑到了，不过我觉得你处理还可以更加优化一些，DNSMASQ 在同时存在ipv4和ipv6地址的情况下，去除ipv6的地址没问题，但是当上游查询结果仅有AAAA记录的情况下应该不进行过滤，这样可以保证日常都是优先ip4，但是纯ipv6的域名也应该能够正确访问。

由于默认情况下的dns查询总是在发起A之后再发起AAAA，所以考虑在DNS的AAAA查询过程中，先去DNS缓存中查询一下是否有A类地址，假如有的话就返回null，否则继续ipv6的正常查询路径

下面的测试代码经过简单测试应该可行，同时建议把“禁止解释 IPv6 DNS 记录”这个选项，改成“IPV4优先”

//patch to filter aaaa forwards
      if (qtype == T_AAAA && option_bool(OPT_FILTER_AAAA) ){
		//查询一次ipv4缓存
	struct crec * queryIpv4 = cache_find_by_name(NULL, name, now, F_IPV4);
	if(queryIpv4 && !(queryIpv4->flags & F_NEG)){
		ans = 1;
		if (!dryrun) log_query(F_CONFIG | F_IPV6 | F_NEG, name, &addr, NULL);
			break;
	}
      }
      //end of patch

wgedu, 2020-02-26:
这段代码加在哪个文件？

coolsnowwolf, 2020-02-26:
请提交一个 PR 比较好

Aspiriniii, 2020-02-27:
PR咋又撤了

coolsnowwolf, 2020-02-27:
测试这个 PR 出现了问题，偶然会造成 DNSMASQ 运行失败

Her0R0cket, 2020-02-27:
这个想法还是很棒的啊

wybb, 2020-02-27:
> 测试这个 PR 出现了问题，偶然会造成 DNSMASQ 运行失败
昨天简单跑了一下，发现在一些主流浏览器并不像nslookup那样，AAAA的请求包会发出的比较早，本打算把从cache中检查IPv4改为调用一次res_query ipv4的查询，居然发现dnsmasq 的处理模式貌似是设计成单线程！！！在一次查询未结束的过程中后续的查询并不会处理，而导致死锁，之前没仔细看过dnsmasq的源代码，一直以为是多线程处理的，而且选项中也有dns并发选项。要大动dnsmasq的源代码很麻烦，几千行代码没有代码说明，自己对linux下的api也不是很熟，看一遍实在太累。
在不改dnsmasq源码比较好做的办法可能还是在前端做个代理，让dnsmasq跑其他端口，侦听53的tcp和udp，在这个阶段对dnsheader进行ipv6判断来确定是否转发给dnsmasq

qingxiwa, 2020-02-27:
我觉得这个不用这样实现，国内的ipv6路由不会乱跑，倒是ipv6国外的路由很大问题，像经欧洲到美国，把ipv6过滤弄到梯子查询国外dns之后，不影响国内ipv6的访问就没问题了

wybb, 2020-02-28:
> 我觉得这个不用这样实现，国内的ipv6路由不会乱跑，倒是ipv6国外的路由很大问题，像经欧洲到美国，把ipv6过滤弄到梯子查询国外dns之后，不影响国内ipv6的访问就没问题了

可能你没碰到过，我就多次碰到一些站点，因为域名有ipv6的地址但速度却比ipv4慢很多，而目前主流的浏览器都是ipv6优先，除非按照微软的标准去修改注册表，毕竟还要考虑其他设备，所以改注册表这种事情适用范围不够方便。

wybb, 2020-02-28:
> 测试这个 PR 出现了问题，偶然会造成 DNSMASQ 运行失败

我这里献丑写了一个简单框架结构的代码实现dns ipv6判断和转发，不是非常严谨而且只跑了一个udp的dns转发，建立一个独立的小线程跑dnsmasq里面就行了，因为对linux和openwrt本身的熟悉度不够所以不敢写完整提交，不知道对你是不是有所帮助。这种实现方式可以让DNSMASQ跑在一个非标端口，自己把dns请求转发给dnsmasq。

void *dns_proxy()
{
  int socket_server, socket_client; 
  char buf[80];
  struct sockaddr_in sin;

  bzero(&sin,sizeof(sin));
  sin.sin_family=AF_INET;
  sin.sin_addr.s_addr=htonl(INADDR_ANY);
  sin.sin_port=htons(5555);  //创建一个新的侦听端口，也可以考虑直接用53端口，让dnsmasq跑其他端口
  socket_server=socket(AF_INET,SOCK_DGRAM,0);
  bind(socket_server,(struct sockaddr *)&sin, sizeof(sin));
  socket_client = socket(AF_INET,SOCK_DGRAM,0);
  
  res_init();
  _res.retry = 1;
  _res.nscount = 1;
  _res.nsaddr_list[0].sin_addr.s_addr = inet_addr("127.0.0.1"); //设置IPv4的查询指向本地的dnsmasq
  _res.nsaddr_list[0].sin_family = AF_INET;
  _res.nsaddr_list[0].sin_port = htons(53); //指向dnsmasq的端口

  while(1)
  {
      char ua[2048];
      struct sockaddr_in socket_remote;
      int addrLen;
		  addrLen = sizeof(struct sockaddr);
      int len = recvfrom(socket_server, ua, sizeof(ua),0, (struct sockaddr *)&socket_remote, &addrLen);
      if(len > 0)
      {
        
        
        
        struct dns_header *header = (struct dns_header *)ua;
        char qname[2048];
        short qtype;
        extract_request(header, len, qname, &qtype);
        if(qtype == T_AAAA)
        {
          //假如是AAAA的查询，用res_query向本地dnsmasq发起一个相同域名的ipv4查询
          //因为是本地查询并且dnsmasq本身就有缓存，所以这个对性能的损失可以忽略不计
          int ipV4count = 0;
          char queryData[2048];
          
          int queryResultLen = res_query(qname, ns_c_in, ns_t_a, queryData, 2048);
          if(queryResultLen)
          {
            ns_msg ns;
            ns_initparse(queryData, queryResultLen, &ns);
            ipV4count = ns_msg_count(ns, ns_s_an);
          }
          syslog(LOG_INFO, "Proxy-> %s:%d Query %s, type=%d, ipv4 result=%d", 
            (char *)inet_ntoa(socket_remote.sin_addr), 
            htons(socket_remote.sin_port),
             qname, qtype, hasIpv4);
          //假如存在ipv4的地址则简单的向查询客户端发送空的查询结果
          if(ipV4count)
          {
            //这里应该发送NO_DATA的reply
            continue;
          }
        }
        //假如不是AAAA查询，那么简单的向本地dnsmasq转发这个请求
        struct sockaddr_in local_server;
 
        bzero(&local_server,sizeof(local_server));
        local_server.sin_family=AF_INET;
        local_server.sin_addr.s_addr=inet_addr("127.0.0.1"); //dnsmasq的本地地址
        local_server.sin_port=htons(53);//dnsmasq的本地端口
        addrLen = sizeof(struct sockaddr);
        sendto(socket_client, ua, len, 0, (struct sockaddr *)&local_server, addrLen);
			  len = recvfrom(socket_client, ua, sizeof(ua), 0, 0, 0);
        if(len > 1)
        {
          sendto(socket_server, ua, len, 0, (struct sockaddr *)&socket_remote, addrLen);
        }
        else
        {
          return 0;
        }
      }
      else
      {
        syslog(LOG_INFO, "Proxy Exit %d", len);
        return 0;
      }

  }
}

qingxiwa, 2020-02-28:
> > 我觉得这个不用这样实现，国内的ipv6路由不会乱跑，倒是ipv6国外的路由很大问题，像经欧洲到美国，把ipv6过滤弄到梯子查询国外dns之后，不影响国内ipv6的访问就没问题了
> 
> 可能你没碰到过，我就多次碰到一些站点，因为域名有ipv6的地址但速度却比ipv4慢很多，而目前主流的浏览器都是ipv6优先，除非按照微软的标准去修改注册表调整优先级，但毕竟还要考虑其他设备，所以改注册表这种事情适用范围也不够方便。

你可以tracert下ipv6 路由怎么跑的，我现在使用smartdns不开ipv6和双栈优选，不影响ipv6访问，好像接近ipv4优先（在这里提到smartdns会不会被骂？），能搞出基于dnsmasq的ipv4优先，我觉得会适合很多人的

closed 2020-03-05 by wybb

jdjingdian, 2020-03-15:
lean大可以完善一下这个功能吗？平常github action编译出来的固件下载贼慢，把IPV6解析禁掉就飞速下载了，但又影响到了IPV6 DDNS的解析，平时有通过IPV6 DDNS访问家里设备的需求，禁止IPV6解析确实对平时上网有帮助，但是ipv6的ddns就用不了了，希望可以增加双栈优选的功能。谢谢！