coolsnowwolf/lede #3356: 导入trojan节点失败

cooguyan, 2020-02-26:
uri是这个：
```
trojan://****@as1-1.ttts-api.xyz:443
```

确定导入后一点反应也没有，我看了 [#3175](https://github.com/coolsnowwolf/lede/issues/3175) 但应该不是因为链接不符合标准。

网页端的报错是这个：

```
Uncaught DOMException: Failed to execute 'atob' on 'Window': The string to be decoded is not correctly encoded.
    at b64decsafe (http://192.168.1.2/cgi-bin/luci/admin/services/shadowsocksr/servers/cfg094a8f:145:16)
    at import_ssr_url (http://192.168.1.2/cgi-bin/luci/admin/services/shadowsocksr/servers/cfg094a8f:287:24)
    at HTMLInputElement.onclick (http://192.168.1.2/cgi-bin/luci/admin/services/shadowsocksr/servers/cfg094a8f:330:144)

```

hxlls, 2020-02-26:
换浏览器试试

coolsnowwolf, 2020-02-26:
你这个trojan 后面并没有加密，不符合标准

SharerMax, 2020-02-26:
> uri是这个：
>
> ```
> trojan://****@as1-1.ttts-api.xyz:443
> ```
>
> 确定导入后一点反应也没有，我看了 [#3175](https://github.com/coolsnowwolf/lede/issues/3175) 但应该不是因为链接不符合标准。
>
> 网页端的报错是这个：
>
> ```
> Uncaught DOMException: Failed to [...]

提供一下版本号

cooguyan, 2020-02-26:
@SharerMax 版本是：1-132

SharerMax, 2020-02-26:
> @SharerMax 版本是：1-132

你该升级了，这个问题已经在 #3195 修复

cooguyan, 2020-02-26:
@SharerMax  谢谢，刚刚安装好新版，现在已经没问题了。我发issue之前看\#3195 了，看你写的是增加了别名支持而我的链接里根本不包含别名,所以就觉得根本用不着更新, 毕竟换了新环境编译一次挺慢的:-D

closed 2020-02-26 by cooguyan