coolsnowwolf/lede #3358: 关于新版 解锁网易云音乐免费白嫖 FLAC 无损音乐的说明

coolsnowwolf, 2020-02-26:
1. 必须更新 feeds 后编译，Node 必须为 10+ 版本，目前已经升级到 v12.15
2. FLAC 音源目前只支持 酷我 和 咪咕 ，但 咪咕  的匹配歌名经常出错
3. 这个版本的 Node 比较吃内存，路由器内存剩余少于 128MB+ 的路由器慎重。。。

PS：无损就是好

[image]

[image]

lphgor, 2020-02-26:
可以更新一下go版，也支持酷我无损了

wspzf, 2020-02-26:
解锁网易云音乐有三个插件，用哪个呢？
现在默认是mini

shu0802, 2020-02-26:
ios手机端支持效果如何

mjyhj, 2020-02-26:
> 解锁网易云音乐有三个插件，用哪个呢？
> 现在默认是mini

标准版是本地node解锁
GO版是适合K3这种不支持FPU的
MINI版适合8M,16M这种小固件的，在大佬上的公益服务器解锁，本地代理过去

zgs360, 2020-02-27:
怎么使用啊？windwos客户端端口设置多少？

xxp2277, 2020-02-27:
@coolsnowwolf 请问有办法将所有歌曲（即网易云音乐中未变灰的低音质歌曲）替换为无损么？

4593645, 2020-02-28:
请问那个menuconfig 选项是解锁网易云？

xxp2277, 2020-02-29:
> 
> 
> 请问那个menuconfig 选项是解锁网易云？

这3个都是luci-app-unblockmusic luci-app-unblockneteasemusic-go luci-app-unblockneteasemusic-mini

closed 2020-03-06 by coolsnowwolf