coolsnowwolf/lede #3359: 关于新版 SSR Plus+ 新加入 Trojan 多线程的支持性能飙升的说明

coolsnowwolf, 2020-02-26:
1. 新版 SSR Plus+ 增加了对 Trojan 多线（进）程并发转发的功能，随着 CPU 内核数的增加，加解密性能几乎线性增长，同等硬件下转发性能大幅超越 V2RAY （服务器带宽无瓶颈的情况下）

2. 为AMD Ryzen 3990x 的 128 线程增加了支持，现在最大支持 128 线程并发转发

3. 希望大家可以在同一个节点测速，回馈下结果。应该会惊喜的。

[image]

hcym, 2020-02-26:
逼我重新部署😂

dolphinpaopao, 2020-02-26:
3990x在招手。。。。。

9l, 2020-02-26:
三万的CPU的做路由器

shu0802, 2020-02-26:
跟上大雕的脚步

swxk521, 2020-02-26:
一般线路VPS 速度trojan和v2ray ws caddy没多大区别，x86 J3160CPU

mcgillkwok, 2020-02-27:
请教一个问题，我的vps是美区服务器，我使用了dns转移，在代理软件可以正常观看亚洲的netflix，但是用ssrplus代理给客户端使用的时候并不能正常观看，有什么解决办法吗？

dengchan0416, 2020-02-27:
Trojan 多线程真的很不错,终于可以跑满200M,显卡不怎么行有点卡,性能飙升得明显
[image]

nathanxie, 2020-02-27:
大佬，你好，我昨天软路由更新固件，用SSR Plus +添加Trojan节点遇到以下问题，请麻烦帮我看下。
1、连接Trojan节点后，谷歌搜索无法打开。
2、油管首页能正常打开，并能正常搜索视频内容，但是点击播放跳转到播放页面也一直无法正常播放，我查看视频速度（详细统计信息），显示没有速度。
3、我通过SSR Plus +设置全局模式进行网页测速和fast测试的时候，均能跑满我的300m宽带，测试ip也都是我vps的ip。
4、我用了3个种模式，gfw和绕过大陆，还有全局模式。只有全局模式能正常测速，以上1，2点还是存在。
Trojan节点在pc客户端和手机客户端均未出现以上问题，请大佬指点下。

SharerMax, 2020-02-27:
> 大佬，你好，我昨天软路由更新固件，用 SSR Plus + 添加 Trojan 节点遇到以下问题，请麻烦帮我看下。
> 1、连接 Trojan 节点后，谷歌搜索无法打开。
> 2、油管首页能正常打开，并能正常搜索视频内容，但是点击播放跳转到播放页面也一直无法正常播放，我查看视频速度（详细统计信息），显示没有速度。
> 3、我通过 SSR Plus + 设置全局模式进行网页测速和 fast [...]

ssr plus使用默认的dns服务，清理dns缓存后重新尝试。

abclai, 2020-02-27:
> 请教一个问题，我的vps是美区，我使用了dns转移，在代理软件可以正常观看亚洲的netflix，但是用ssrplus代理给客户端使用的时候并不能正常观看，有什么解决办法吗？

我也遇到这个问题，按DNS解锁提供方的说法是现在路由器的有时候有伪装/混淆网站，还有本身酸酸乳就重新定位了DNS(比如8.8.8.8)变成传输到DNS解锁方的时候DNS信息已经发生变化！所以现在一般只包手机/PC客户端

SharerMax, 2020-02-27:
> > 请教一个问题，我的 vps 是美区，我使用了 dns 转移，在代理软件可以正常观看亚洲的 netflix，但是用 ssrplus 代理给客户端使用的时候并不能正常观看，有什么解决办法吗？
> 
> 我也遇到这个问题，按 DNS 解锁提供方的说法是现在路由器的有时候有伪装 / 混淆网站，还有本身酸酸乳就重新定位了 DNS (比如 8.8.8.8) 变成传输到 DNS 解锁方的时候 DNS 信息已经发生变化！所以现在一般只包手机 / PC 客户端

#3230 
可尝试自定义dns

ps：非相关标题的issue建议另开issue

AmibaWorld, 2020-02-27:
> Trojan 多线程真的很不错,终于可以跑满200M,显卡不怎么行有点卡,性能飙升得明显,
> 多线程,线程数选自动,CPU 3710是4线程
> [image]
> 
> 单线程
> [image]

请问你的服务器在哪里啊，速度好快啊

preangelleo, 2020-02-28:
我用了一周多了，路由器端我一直选用8线程的，我的菜单里有CPU线程数自动，但是没有超过8线程的选择。可能是我的lede版本问题。不过trojan用下来一直比v2ray快一倍不止，只不过我的trojan是搭在google vps的，V2Ray是别的机场的，费用不同，所以我平时都用便宜的v2ray，不稳定的时候切换到”按流量付费的GOOGLE上“，以前在google vps也搭了v2ray，跑下来还是不如trojan稳定和高速，所以就删除了。

bdhjiajia, 2020-02-28:
所以看起来服务器端需要重新部署哇，之前用的一键脚本也会自动更新吗？

467815891a, 2020-02-28:
真的奇怪，我完全感觉不到速度增加。服务器端的"reuse_port"需要开吗？我的7621路由器只能跑8Mbps到头了

hcym, 2020-02-28:
开了32，有效果，就怕流量大被钉钉🧐🙄

jiayouzl, 2020-02-28:
这个真是好东西j1900多线程能力不错,雄起了!~

404user, 2020-02-28:
> 这个真是好东西j1900多线程能力不错,雄起了!~

同U，请问你效果明显吗？

hcym, 2020-02-28:
主要还是看线路，🤭

KEVINJ1E, 2020-02-28:
几乎跑满宽带带宽

404user, 2020-02-28:
>主要还是看线路，🤭

@hcym 我用的是gcp……

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

dengchan0416, 2020-02-28:
@AmibaWorld 香港节点

coolsnowwolf, 2020-02-28:
新增了 Trojan 订阅支持了

404user, 2020-02-29:
L大，对比了一下不知道为什么感觉Trojan 速度比V2RAY慢，尤其在Instagram加载的速度上慢，同一节点，南方电信。

hcym, 2020-02-29:
慢是肯定不慢，刚刚开始用估计运营商还没QoS，比V2快不少，现在也没有什么感觉了，但是CPU肯定低多了，稳定性也好一些

coolsnowwolf, 2020-02-29:
PS：在 MIPS 或者 K3 这种不带 FPU 的 CPU 上面，因为 FPU 是编译器或者内核模拟的，浮点在内部还是单线程状态，提升可能不会明显，或者没有线性提升

推荐用带 FPU 的 ARM，或者直接 x86

ch1993128, 2020-03-01:
用的mtk mips 7621.并没有什么感觉，有机会拿树莓派测试一下

xinpenghan, 2020-03-01:
> ssr plus使用默认的dns服务，清理dns缓存后重新尝试。

我的也是，昨天刚编译的，谷歌打不开，youtube正常，但是过会又没速度了，现在换回原来的固件恢复正常

jiayouzl, 2020-03-01:
Trojan忽然没速度我也碰到了.最新固件

xinpenghan, 2020-03-02:
> Trojan忽然没速度我也碰到了.最新固件

我用的是ssr，刚开始以为配置错了，后来又重刷了固件，重新设置了几次，还是一样youtube可以打开，谷歌过一会就打不开了，youtube速度很慢，换回原来的固件正常…

mianmianmi, 2020-03-02:
大神们，你们好。我想请教关于Trojan的一个问题。我用vultr的VPS（东京的）用脚本做了一个VPN。但是很不稳定。有时候连480P的视频看着都卡。我选的还是一颗CPU，2G内存2000G流量的VPS。完全蒙蔽啊。不知道为什么会慢。用IPIP.NET查IP的PING延迟还可以。113MS。哎。。。。有大神告诉我是怎么回事吗？

hcym, 2020-03-02:
vultr现在对中国的线路不行了，ping没意义的，带宽和丢包才要命，我的是移动宽带，去年上半年以前用vultr的ipv6都能跑满100的带宽，用欧洲节点看4k也稳定的，六月以后就不行了，今年是完全木速度，昨天也开了一个节点，ping值不超过95ms，速度勉强1080，和瓦工ping值200以上的没法子比，他们家机器是真的好，就是没有GIA的

winfylee, 2020-03-03:
开了4线程，对比测试下来电脑上油管连接速度至少翻了倍，从原先的25000到了50000多甚至75000。但奇怪的是，电脑上的效果很明显，手机上效果反而有所下降，同样网络环境下，手机连油管时快时慢，连telegram则速度一直不理想，不知道是不是个案😂😂😂

467815891a, 2020-03-03:
我也有这样的感觉，看视频速度上去了，tg连接的速度变慢了

norikohxb, 2020-03-04:
客户端如果开了128线程，谷歌的那个gcp要不要做什么相应设置？

guochan2019, 2020-03-04:
> 大神们，你们好。我想请教关于Trojan的一个问题。我用vultr的VPS（东京的）用脚本做了一个VPN。但是很不稳定。有时候连480P的视频看着都卡。我选的还是一颗CPU，2G内存2000G流量的VPS。完全蒙蔽啊。不知道为什么会慢。用IPIP.NET查IP的PING延迟还可以。113MS。哎。。。。有大神告诉我是怎么回事吗？

依个人经验，一是vultr的登录帐户验证要改成证书，否则每天不少于几百次的攻击型登录请求，直接影响你的带宽；二是vultr最好用搭建V2Ray，Trojan日常性断流会让你崩溃，我也不知为什么，经验；三是vultr对联通很友好,自己搭的V2Ray,白天油管3万左右,晚上8万附近,好的时候10几万,移动和电信估计就不会太乐观了.

Wilson-coder, 2020-03-04:
同一个Vultr节点，同时开通了两个VPS，一个部署的是V2ray，Websocket+TLS，一直在用，已经升级到了最新版；另一个用的是Trajon，今天部署的。测试下来除了用fast.com网站差别较大之外，看油管发觉差异非常小，也许Trajon速度快一些，但是不排除心理作用。平台是3205U的软路由，默认开的是8线程，网络是深圳电信300M宽带。改成16线程好像比8线程快一些，但是到了32线程好像还降速了，也许是U太弱了。

zxzxvic, 2020-03-04:
> 我的也是，昨天刚编译的，谷歌打不开，youtube正常，但是过会又没速度了，现在换回原来的固件恢复正常

我的也是，一键布置trojan到GCP的，使用SSR+，谷歌打不开，youtube能打开，无法播放视频，折腾了一晚上了，求解

norikohxb, 2020-03-05:
lean大你自己是不是用AMD Ryzen 3990x 的 128 做软路由啊？

ch1993128, 2020-03-06:
彷佛在逗我。。。。。128线程。。。。建机场用嘛。。。

touzi, 2020-03-06:
给个单独下载地址

closed 2020-03-06 by coolsnowwolf

ncxukc, 2020-03-08:
专门来顶大雕的

ssrpanel, 2020-03-17:
> 新增了 Trojan 订阅支持了

订阅格式是什么样子的呀？

mianmianmi, 2020-03-17:
要不要用用我的VPN试试

Neverwintern, 2020-05-02:
It seems this package was deleted. I have searched the entire internet only found an old version backup with no trojan support. 

 Could you please share a repository for the new SSR plus +?  Thank you!

norikohxb, 2020-05-02:
> It seems this package was deleted. I have searched the entire internet only found an old version backup with no trojan support.
> 
> Could you please share a repository for the new SSR plus +? Thank you!

https://github.com/fw876/helloworld

WROIATE, 2020-06-03:
> It seems this package was deleted. I have searched the entire internet only found an old version backup with no trojan support.
> 
> Could you please share a repository for the new SSR plus +? Thank you!

you need to modify the feeds with helloworld

467815891a, 2020-06-08:
> It seems this package was deleted. I have searched the entire internet only found an old version backup with no trojan support.
> 
> Could you please share a repository for the new SSR plus +? Thank you!

uncomment the last line of feeds.conf.default

dengchan0416, 2020-06-09:
> > 似乎此软件包已删除。我搜索了整个互联网，但只发现了一个没有特洛伊木马程序支持的旧版本备份。
> > 您能否为新的SSR plus +共享存储库？谢谢！
> 
> 取消注释feeds.conf.default的最后一行

https://github.com/fw876/helloworld

ztfot, 2020-10-03:
> 1. 新版 SSR Plus+ 增加了对 Trojan 多线（进）程并发转发的功能，随着 CPU 内核数的增加，加解密性能几乎线性增长，同等硬件下转发性能大幅超越 V2RAY （服务器带宽无瓶颈的情况下）
> 2. 为AMD Ryzen 3990x 的 128 线程增加了支持，现在最大支持 128 线程并发转发
> 3. 希望大家可以在同一个节点测速，回馈下结果。应该会惊喜的。
> 
> [image]

好像使用ssrplus+的trojan节点打不开http://trojan-tutor.github.io/

wbpluto, 2023-02-10:
> 真的奇怪，我完全感觉不到速度增加。服务器端的"reuse_port"需要开吗？我的7621路由器只能跑8Mbps到头了

7621路由器，跑满本地宽带的80%

mianmianmi, 2023-02-10:
是不是发错邮件了？？