coolsnowwolf/lede #3364: uPnP运行不正常

xiaoqingfengATGH, 2020-02-26:
MiniuPnPd运行不正常，
OpenWrt 19.07原版正常。

现象，支持upnp软件无法发现路由器，也无法进行端口映射。

hxlls, 2020-02-26:
[image]
s实测正常，请自己检查

xiaoqingfengATGH, 2020-02-26:
Upnp已经不好用很久了，之前拿到了几个lede版本，刷入路由器，发现BitComet，EMule报找不到upnp路由错误。

最近一次验证 2020.2.26 lede master代码，自行编译。
[image]

IP：192.168.5.247 开启upnp
[image]

同网段启动一台原版OpenWrt 19.07，IP：192.168.5.252 开启upnp
[image]

同网段Windows10 使用upnpc进行搜索，只能搜索到192.168.5.252，找不到LEDE的 192.168.5.247
[image]
如果关闭原版OpenWrt 19.07，执行upnpc，会无法搜索到upnp设备
[image]

coolsnowwolf, 2020-02-26:
这个是个选择题

因为只有一个版本的 UPNPD 兼容迅雷所有版本

如果选了官方的 UPNPD，那么迅雷就各种异常，2选1 ，你们觉得呢

coolsnowwolf, 2020-02-26:
另外，编译了 MWAN3 后 UPNP 会失效，卸载就好了

[image]

zhanghao1989, 2020-02-26:
放两个upnp 😀 一个官方，一个迅雷。

ntlf9t, 2020-02-26:
多线用户或者多lan口都会造成upnp失效需要自己改配置文件

xiaoqingfengATGH, 2020-02-27:
@ntf9t 您的意思是修改配置文件可以使得MWAN3和 upnpd同时生效？能解释下修改方法吗？

xiaoqingfengATGH, 2020-02-27:
> 另外，编译了 MWAN3 后 UPNP 会失效，卸载就好了
> 
> [image]

有一个其他情况，编译了Lienol维护的OpenWrt，编译时，同时加入了MWAN3和miniUpnpd，使用相同的测试方法，内网IP 192.168.5.247
[image]
[image]

没有修改过任何配置文件，测试upnp有效。
[image]

Lienol 版本内核低于目前master的。
[image]

xiaoqingfengATGH, 2020-02-27:
> 多线用户或者多lan口都会造成upnp失效需要自己改配置文件

补充一下，昨天测试使用的官方OpenWrt 19.07版本也同时安装了mwan3和miniupnpd。从测试结果来看，运行正常。

[image]
[image]

coolsnowwolf, 2020-02-27:
那就去掉支持迅雷的那个版本？ 反正我也懒得支持了

xiaoqingfengATGH, 2020-02-27:
看了下运行正常的Lienol的miniunpnpd和lede的miniupnpd，版本不同
左侧是Lienol右侧是lede
[image]

xiaoqingfengATGH, 2020-02-27:
> 那就去掉支持迅雷的那个版本？ 反正我也懒得支持了

两个共存，可行吗？

coolsnowwolf, 2020-02-27:
已更换，as your wish

[image]

迅雷全家桶用不了 UPNP的，勾一下 v1 看看是否有效

不能共存

knightridersimon, 2020-02-27:
wan3+upnp, 迅雷54321等端口正常，但是群辉检测不行。

coolsnowwolf, 2020-02-27:
2 选一

closed 2020-02-27 by coolsnowwolf

xiaoqingfengATGH, 2020-02-27:
经过测试只将UPNP升级是无效的。

请同时将MWAN3也升级，即可共存。
MWAN3版本如下：
[image]

xhqpp, 2020-03-21:
> @ntf9t 您的意思是修改配置文件可以使得MWAN3和 upnpd同时生效？能解释下修改方法吗？

请问多拨负载均衡和upnp同时生效的方法能说下吗？