coolsnowwolf/lede #3367: Ubuntu18.4 下载依赖文件出错 路由器绕过大陆ip模式

RA089, 2020-02-26:
kang@kang-pc:~$ sudo apt-get update
[sudo] password for kang:
Hit:1 http://archive.ubuntu.com/ubuntu bionic InRelease
Hit:2 http://archive.ubuntu.com/ubuntu bionic-updates InRelease
Hit:3 http://security.ubuntu.com/ubuntu bionic-security InRelease
Hit:4 http://archive.ubuntu.com/ubuntu bionic-backports InRelease
Reading package lists... Done
kang@kang-pc:~$ sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3.5 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib linux-libc-dev:i386
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'zlib1g-dev' instead of 'libz-dev'
Note, selecting 'libpython3.5-minimal' for regex 'python3.5'
Note, selecting 'node-uglify' instead of 'uglifyjs'
Note, selecting 'git' instead of 'git-core'
Note, selecting 'upx-ucl' instead of 'upx'
E: Unable to locate package linux-libc-dev:i386
kang@kang-pc:~$

bin20088, 2020-02-26:
遇到同样的问题，18.04

AmadeusGhost, 2020-02-27:
有一些是14.04的包，18.04用这个
```
sudo apt-get install -y build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib g++-multilib p7zip p7zip-full msmtp libssl-dev texinfo libreadline-dev libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint ccache curl wget scons python python3 python-pip python3-pip python-ply python3-ply haveged lrzsz device-tree-compiler
```

you-mu, 2020-02-27:
供参考:
编译的环境是18.04 ubuntu server版, 关闭ipv6, LEDE设置这台全局代理.

naoki66, 2020-02-27:
我是失败在这里
Building targets
go: github.com/v2ray/v2ray-core@v4.22.1+incompatible: reading https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod: 404 Not Found

应该是go问题 但是我没有能力解决

AmadeusGhost, 2020-02-27:
> 我是失败在这里
> Building targets
> go: github.com/v2ray/v2ray-core@v4.22.1+incompatible: reading https://goproxy.cn/github.com/v2ray/v2ray-core/@v/v4.22.1+incompatible.mod: 404 Not Found
> 
> 应该是go问题 但是我没有能力解决

输入`go env -w "GOPROXY="`

naoki66, 2020-02-27:
alex@ubuntu18:~/lede$ sudo go env -w GOPROXY=https://goproxy.cn,direct
问题依旧

closed 2020-02-27 by coolsnowwolf

blizzardster, 2020-03-09:
> apt-get install -y build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib [...]

好用，解决了，谢谢分享